clash-lang/clash-compiler #46: Initial blocks vs reset line

Ericson2314, 2015-03-09:
Quartus + Cyclone II for example supports synthesizing initial blocks--or at least those that initialize registers with a literal (that's what I've tried). It can also do this more efficiently than with a reset line in some cases.

Perhaps an option should be added for CLaSH to use a register primitive with initial blocks, and forgo adding extra reset lines?

christiaanb, 2015-03-09:
This can only work if the reset value is a constant. The [blockRam](https://github.com/christiaanb/clash-prelude/blob/master/src/CLaSH/Prelude/BlockRam.hs#L76) primitive already demands that it's initial value must be a constant in its documentation, but it is not enforced in the front-end of the compiler. Indeed, compilation will only fail once it is already in the code-generation phase.

At the moment, there is no such demand for registers _because_ it is currently initialised through a reset. Perhaps what we can do is use an initial block when the reset value is a constant, something we check at code-generation time.

Ericson2314, 2015-03-10:
Hmm, in a synthesizable program all values should be either a constant or a signal, unless it has `topEntity :: A -> Signal B` (with no signals in A), right?

christiaanb, 2015-03-11:
Yeah, that use-case makes no sense. Indeed, the reset value should _ultimately_ be a constant.

On the implementation side, we sometimes end up with code like this:
```haskell
topEntity x y = ...
  where
    (c,d) = g k p
    e     = f c x
    ...

g k p = (3,k)

f d x = register d x
```
Where the reset value of `register` is indeed a constant, but the `register` function is not _applied_ to a constant. In the current setup for the compiler, and forcing the reset value to be a literal, compilation of the above code will fail at code-generation.

What needs to happen, is that the compiler looks at all the BlackBox definitions and check which primitives need literals for certain arguments. Subsequently, the normalisation phase should try it's hardest to make sure literals end up at those places. The above code would then probably be transformed first to:
```haskell
topEntity x y = ...
  where
    (c,d) = (3,k)
    e     = f c x
    ...

f d x = register d x
```
and then to:
```haskell
topEntity x y = ...
  where
    d     = k
    e     = f x
    ...

f x = register 3 x
```
Anyhow, it's not as trivial as it might have initially sounded.

Ericson2314, 2015-03-11:
Yeah, definitely not so trivial then unfortunately. Pity `https://ghc.haskell.org/trac/ghc/wiki/Supercompilation` never got merged. Even if it is not good in the general case, I assume it would be really useful for CLaSH.

christiaanb, 2015-03-14:
I have a collegae that went with a supercompilation approach described at: http://www.cs.ru.nl/P.Achten/IFL2013/symposium_proceedings_IFL2013/ifl2013_submission_7.pdf

The problem with supercompilation for me, is that it completely destroys the first-order function hierarchy. I think a nice aspect of CLaSH's current approach, is that the resulting netlist looks very much like the original function hierarchy. I aids in both the understanding of how this "structural" approach to designing (in Haskell) works, and it aids in debugging non-functional aspects of your design (propagation delays and area). That is, it is nice to be able to look at your RTL diagram, and see where your design is using a lot of area, or incurring a lot of propagation delay, and then immideately figure out which function is causing that.

When you destroy the original first-order function hierarchy, those things become (much) harder.

Ericson2314, 2015-04-30:
Interesting paper (and sorry I didn't respond earlier--I read this a while back and then forgot to respond).

Good points on legibility post compilation. Taking the opposite tack, I wonder if using macro-parametrized modules could avoid needing monomorphization. Consider this module I wrote:
```systemverilog
module BCD #( parameter BITS_IN
            , parameter DIGITS)
            ( input [BITS_IN-1:0] binary
            , output reg [DIGITS-1:0][3:0] digits);
  always_comb @(*) begin
    digits = 1'd0;
    
    for (integer i=BITS_IN; i>=0; i=i-1) begin
      // Add 3 to columns that are >= 5
      for (integer j=DIGITS-1; j>=0; j=j-1) begin
        if (digits[j] >= 5)
          digits[j] = digits[j] + 4'd3;
      end
      digits = {digits, binary[i]};
    end
  end
endmodule
```
[Finally, I am not sure how much a good idea my original proposal is even if it was feasible--hooking up the PLL lock to the reset-line is still needed even if the initial value is a constant.]

martijnbastiaan, 2019-02-15:
We've added register powerup values (or initial blocks) in #498. Any argument used with `~CONST` (in a blackbox) will make Clash try extra hard to reduce it to a constant. We've decided not to not explicitely support registers without reset values, to prevent a zoo of functions. However, you could try and use `Clash.Signal.Internal.register#`, which allows you to specify a different reset value. HDL tools might be smart enough to completely let go of reset logic if you pass `undefined` there. Alternatively, you could create your own register primitive. For inspiration you can find the current implementation [over here](https://github.com/clash-lang/clash-compiler/blob/master/clash-lib/prims/vhdl/Clash_Signal_Internal.json#L53-L141).

I believe the PR fixes this issue, so I'm closing this issue. If that's a mistake, please reopen!

closed 2019-02-15 by martijnbastiaan