clash-lang/clash-compiler #50: GHC 7.10 compatibility?

mgajda, 2015-04-18:
Since [GHC 7.10](https://www.haskell.org/ghc/download_ghc_7_10_1) is out yet, it would be great to add compatibility for this version.

I'd also love to see clash-ghc and clash-prelude to become part of [Stackage](https://www.stackage.org/), and thus automatically checked against all compiler releases supported by [Stackage](https://www.stackage.org/).

christiaanb, 2015-04-18:
I'm working on it. See  https://github.com/christiaanb/clash-prelude/tree/ghc-typelits-natnormalise and https://github.com/christiaanb/clash-compiler/tree/ghc-7.10. I'll release once I get all the tests for clash-prelude working. Right now they are still failing.

christiaanb, 2015-04-20:
http://hackage.haskell.org/package/clash-ghc-0.5.1, which supports, and only works on, ghc 7.10, is now released.

closed 2015-04-20 by christiaanb

mgajda, 2015-04-20:
Is there matching version of clash-prelude?

christiaanb, 2015-04-20:
Yes: http://hackage.haskell.org/package/clash-prelude-0.7.2