clash-lang/clash-compiler #53: Lost system1000 clock in VHDL generation...

mgajda, 2015-05-01:
CLaSH compiler occasionally _misses_ the clock forwarding to an entity:
```
entity topEntity28_14 is
  port(eta_i1   : in unsigned(10 downto 0);
       topLet_o : out product5);
end;

architecture structural of topEntity28_14 is
```
...and then it tries to instantiate subordinate entity with this missing clock:
```
  mealyzm_7_tup_1 : entity mealyzm_7
    port mapl
      (bodyVar_o      => tup_1
      ,system1000     => system1000
      ,system1000_rst => system1000_rst);
```
That naturally leads to uncompilable VHDL code.

christiaanb, 2015-05-01:
Is it possible to provide a test case?

mgajda, 2015-05-01:
Yes, I provided it in lPaste link.

mgajda, 2015-05-01:
PS I might provide patch to treat it as an example for CLaSH, when it will be able to generate correct VHDL.

closed 2015-05-03 by christiaanb

christiaanb, 2015-05-05:
Bugfix is included in version 0.5.5 that I just released.

commit 234ee14d (2015-05-03): 'tell' hidden ports for instantiated component

The problem that arose was that a component had its hidden ports
mapped, but the encompasssing component didn't have corresponding
hidden ports.

This happened because we were cashing generated components, and
the list of hidden ports wouldn't be added to the Writer monad.

Fixes #53