clash-lang/clash-compiler #55: VHDL syntax error when generating array of records...

mgajda, 2015-05-09:
Latest CLaSH 0.5 generates invalid VHDL, according to Xilinx ISE compiler:
```
Compiling vhdl file "/home/m/exp/fpga/vga/vhdl/Main/topEntity1_11.vhdl" in Library work.
Entity <topentity1_11> compiled.
ERROR:HDLParsers:164 - "/home/m/exp/fpga/vga/vhdl/Main/topEntity1_11.vhdl" Line 27. parse error, unexpected SEMICOLON, expecting COMMA or CLOSEPAR
```
Relevant VHDL code fragment is as follows:
```
    n_32 <= array_of_product1'((product1_sel0 => unsigned'(3-1 downto 0 => '1');,product1_sel1 => unsigned'(3-1 downto 0 => '0');,product1_sel2 => unsigned'(3-1 downto 0 => '0');),(product1_sel0 => unsigned'(3-1 downto 0 => '1');,product1_sel1 => unsigned'(3-1 downto 0 => '1');,product1_sel2 => unsigned'(3-1 downto 0 => '0');),(product1_sel0 => unsigned'(3-1 downto 0 => '0');,product1_sel1 => unsigned'(3-1 downto 0 => '1');,product1_sel2 => unsigned'(3-1 downto 0 => '0');),(product1_sel0 => unsigned'(3-1 downto 0 => '0');,product1_sel1 => unsigned'(3-1 downto 0 => '0');,product1_sel2 => unsigned'(3-1 downto 0 => '1');),(product1_sel0 => unsigned'(3-1 downto 0 => '1');,product1_sel1 => unsigned'(3-1 downto 0 => '0');,product1_sel2 => unsigned'(3-1 downto 0 => '1');),(product1_sel0 => unsigned'(3-1 downto 0 => '0');,product1_sel1 => unsigned'(3-1 downto 0 => '1');,product1_sel2 => unsigned'(3-1 downto 0 => '1');),(product1_sel0 => unsigned'(3-1 downto 0 => '1');,product1_sel1 => unsigned'(3-1 downto 0 => '1');,product1_sel2 => unsigned'(3-1 downto 0 => '1');),(product1_sel0 => unsigned'(3-1 downto 0 => '0');,product1_sel1 => unsigned'(3-1 downto 0 => '0');,product1_sel2 => unsigned'(3-1 downto 0 => '0');));
    n_33 <= repANF_1;
```

It seems from creation of `Vec`tor of records:
```haskell
        colors     :: Vec 8 (VGAColor 3)
        colors      = red :> yellow :> green :> blue :> magenta :> cyan :> white :> black :> Nil 
```
I put entire CLaSH input file on [Paste](http://lpaste.net/5990385496994349056)

Even if you plan to revamp source entirely, this seems to be a good candidate to add to the unit tests.

christiaanb, 2015-05-09:
Thanks for the bug-report. I fixed it for 'BitVector', but forgot to check the same error in 'Unsigned'. I just pushed a fix: https://github.com/clash-lang/clash-compiler/commit/3a5d5ad415c88b37dd1d6ccff5e8013620baae94

As a workaround until I push a new package to hackage: you can add the following blackbox definitions to the directory from which you run the clash compile.

`CLaSH.Sized.Internal.Unsigned.json`
```
[ { "BlackBox" :
    { "name"      : "CLaSH.Sized.Internal.Unsigned.minBound#"
    , "templateE" : "unsigned'(~LIT[0]-1 downto 0 => '0')"
    }
  }
, { "BlackBox" :
    { "name"      : "CLaSH.Sized.Internal.Unsigned.maxBound#"
    , "templateE" : "unsigned'(~LIT[0]-1 downto 0 => '1')"
    }
  }
]
```

The CLaSH compiler will then overwrite the default translation rule for minbound and maxbound with the correct definitions.

mgajda, 2015-05-10:
Thanks a lot!

Do you plan a tutorial on making your own `BlackBox`es and emitting own compilation patterns?

closed 2015-05-10 by mgajda

christiaanb, 2015-05-10:
I've just released version 0.5.4 of the VHDL backend. To install:
```
$ cabal update
$ cabal install clash-vhdl-0.5.4 clash-systemverilog-0.5.4 clash-ghc
```