clash-lang/clash-compiler #56: How to define Block ROM BlackBox...

mgajda, 2015-05-10:
Hi Christiaan,

Considering that both _Altera_ and _Xilinx_ support *Block ROM* _IP Core_ generators, and these seem to work quite fast, I'd love to see *Block ROM* black box defined.

Could you please lend me a hand, and give a tutorial about how to define such a custom `BlackBox`? Is it something an average user could do, or is it supposed to be hard-wired in the internals of the compiler?

christiaanb, 2015-05-10:
There are already black boxes for the following functions: http://hackage.haskell.org/package/clash-prelude-0.7.5/docs/CLaSH-Prelude-BlockRam.html

There's also already a section in the tutorial on custom black boxes: http://hackage.haskell.org/package/clash-prelude-0.7.5/docs/CLaSH-Tutorial.html#g:13

Could you elaborate what you feel is missing from the tutorial?

mgajda, 2015-10-12:
Thanks a lot for info!

closed 2015-10-12 by mgajda