clash-lang/clash-compiler #57: Normalization failure with ADT

mgajda, 2015-05-10:
I've just tried to abstract my VGA controller over different possible modeline parameters, and got a rather obscure error message:
```
[1 of 2] Compiling Modeline         ( Modeline.hs, Modeline.o )
[2 of 2] Compiling Main             ( Video.hs, Video.o )
Loading dependencies took 0.473015s

<no location info>:
    CLaSH Error:
CLaSH.Normalize(96): Expr belonging to bndr: nubBy'1272 (:: GHC.Types.[]
  (GHC.Tuple.(,,) GHC.Types.Double GHC.Types.Int GHC.Types.Int)
-> GHC.Tuple.(,,) GHC.Types.Double GHC.Types.Int GHC.Types.Int
-> GHC.Types.[]
     (GHC.Tuple.(,,) GHC.Types.Double GHC.Types.Int GHC.Types.Int)) remains recursive after normalization:
λ(pTS :: GHC.Types.[]
           (GHC.Tuple.(,,) GHC.Types.Double GHC.Types.Int GHC.Types.Int)) ->
λ(y :: GHC.Tuple.(,,)
         GHC.Types.Double GHC.Types.Int GHC.Types.Int) ->
letrec
  subjLet :: GHC.Types.Bool
  = Data.OldList.elem_by
    @(GHC.Tuple.(,,) GHC.Types.Double GHC.Types.Int GHC.Types.Int)
      Modeline.modeline91912625679
      y1
      (GHC.Types.:
       @(GHC.Tuple.(,,) GHC.Types.Double GHC.Types.Int GHC.Types.Int)
         y
         (GHC.Types.[]
          @(GHC.Tuple.(,,) GHC.Types.Double GHC.Types.Int GHC.Types.Int)))
  altLet2 :: GHC.Types.[]
               (GHC.Tuple.(,,) GHC.Types.Double GHC.Types.Int GHC.Types.Int)
  = GHC.Types.:
    @(GHC.Tuple.(,,) GHC.Types.Double GHC.Types.Int GHC.Types.Int)
      y1
      (nubBy'1321
         ys
         y1
         y)
  altLet1 :: GHC.Types.[]
               (GHC.Tuple.(,,) GHC.Types.Double GHC.Types.Int GHC.Types.Int)
  = nubBy'1272
      ys
      y
  altLet :: GHC.Types.[]
              (GHC.Tuple.(,,) GHC.Types.Double GHC.Types.Int GHC.Types.Int)
  = case subjLet of
      GHC.Types.False -> altLet2
      GHC.Types.True -> altLet1
  y1 :: GHC.Tuple.(,,) GHC.Types.Double GHC.Types.Int GHC.Types.Int
  = case pTS of
      GHC.Types.:
        (sel :: GHC.Tuple.(,,)
                  GHC.Types.Double GHC.Types.Int GHC.Types.Int)
        (wild :: GHC.Types.[]
                   (GHC.Tuple.(,,) GHC.Types.Double GHC.Types.Int GHC.Types.Int)) ->
        sel
  ys :: GHC.Types.[]
          (GHC.Tuple.(,,) GHC.Types.Double GHC.Types.Int GHC.Types.Int)
  = case pTS of
      GHC.Types.:
        (wild :: GHC.Tuple.(,,)
                   GHC.Types.Double GHC.Types.Int GHC.Types.Int)
        (sel :: GHC.Types.[]
                  (GHC.Tuple.(,,) GHC.Types.Double GHC.Types.Int GHC.Types.Int)) ->
        sel
in case pTS of
     GHC.Types.[] ->
       GHC.Types.[]
       @(GHC.Tuple.(,,) GHC.Types.Double GHC.Types.Int GHC.Types.Int)
     GHC.Types.:
       (y2 :: GHC.Tuple.(,,) GHC.Types.Double GHC.Types.Int GHC.Types.Int)
       (ys1 :: GHC.Types.[]
        ru     (GHC.Tuple.(,,) GHC.Types.Double GHC.Types.Int GHC.Types.Int)) ->
       altLet
```
Most of the types and indeed most of the code is identical with previous version, just it is abstracted over `Modeline`, I have no idea what to attribute it to.

I attach _lpaste_ links to the two [source](http://lpaste.net/8332104136103297024) [modules](http://lpaste.net/1102915393653047296).

mgajda, 2015-05-10:
It may be a part of more general question: is it possible to mark constants that require some recursive Haskell-level evaluation as constants that need to be evaluated before _CLaSH_ generates VHDL code, or is it necessary to use other tricks here? (Like _Template Haskell_.)

christiaanb, 2015-05-10:
It pains me to say that you'll need to use template Haskell (tricks) here. The compile time evaluation mechanisms are very poor in the clash compiler right now. It is one of the reasons I want to overhaul the internals. Right now you should avoid any use of lists (and 'Double' for that matter) outside of a template Haskell expression.

Ericson2314, 2015-05-10:
Hmm, Enforcing a phase separation with TH between the Haskell CLaSH can deal with and that which it can't doesn't seem like the worst thing to me.

Ericson2314, 2019-09-03:
https://github.com/ghc-proposals/ghc-proposals/pull/243 Years later I finally get around to proposing that phase separation.