clash-lang/clash-compiler #59: demo on the tutorial does not work

Alaya-in-Matrix, 2015-06-02:
I am trying to learn CLaSH following the turorial at http://hackage.haskell.org/package/clash-prelude-0.5/docs/CLaSH-Tutorial.html#g:5

for example, there is no `simulateP`, but there are `simulate` and `simulateB`.

christiaanb, 2015-06-02:
Hi, what version of clash are you using:
```
$ clash --version
```
And what version of the clash-prelude?
```
$ ghc-pkg list | grep clash-prelude
```

The tutorial for `clash-prelude-0.7.5` http://hackage.haskell.org/package/clash-prelude-0.7.5/docs/CLaSH-Tutorial.html#g:5 mentions `simulate`.
And, `clash-prelude-0.5` has [simulateP](http://hackage.haskell.org/package/clash-prelude-0.5/docs/CLaSH-Signal-Implicit.html#v:simulateP)

Alaya-in-Matrix, 2015-06-02:
the version of `clash` and `clash-prelude`:
>CAES Language for Synchronous Hardware, version 0.5.6 (using clash-lib, version: 0.5.5)
    clash-prelude-0.7.5

when runing below command:
```
clash --interactive Mac.hs
```
where the content of `Mac.hs` is

```haskell
module Mac where 
import CLaSH.Prelude 
import CLaSH.Signal.Explicit

ma acc (x,y) = acc + x * y
-- ret = ma 3 4

macT acc (x,y) = (acc',o)
  where acc' = ma acc (x,y)
        o    = acc

mac = macT <^> 0
```

then type `:t simulateP` in the prelude, I got the "Not in scope" error

christiaanb, 2015-06-02:
Yes, `clash-prelude-0.7.5` no longer contains `simulateP`. I don't know what happened, but you somehow have been following an older version of the tutorial.

Please follow the latest version at: http://hackage.haskell.org/package/clash-prelude-0.7.5/docs/CLaSH-Tutorial.html

closed 2015-06-02 by christiaanb

Alaya-in-Matrix, 2015-06-02:
cool... I just used "clash tutorial" as the google keyword and that deprecated hackage tutorial just came out  as the first page google searched for me...

masterdezign, 2015-06-02:
Google does not necessary return the latest versions of the documentation. So you have to pay attention to versions when searching that way.

Ericson2314, 2015-06-02:
You can always strip the version in the Hackage URL, and then it will default to the latest.

masterdezign, 2015-06-02:
Exactly, 

so http://hackage.haskell.org/package/clash-prelude-0.7.5/docs/CLaSH-Tutorial.html
becomes http://hackage.haskell.org/package/clash-prelude/docs/CLaSH-Tutorial.html