clash-lang/clash-compiler #61: Test fails after upgrading to clash-ghc 0.5.8

masterdezign, 2015-06-18:
Hi!

To prevent a potential bug, please see https://github.com/masterdezign/clash-fifo-test/

The test started to fail after upgrade to (yet unreleased) clash-ghc-0.5.8

$ clash --interactive Test/Fifo2
CLaSHi, version 0.5.8 (using clash-lib, version 0.5.7):
http://www.clash-lang.org/  :? for help
[1 of 3] Compiling Clocks           ( Clocks.hs, interpreted )
[2 of 3] Compiling Fifo2            ( Fifo2.hs, interpreted )
[3 of 3] Compiling Test.Fifo2       ( Test/Fifo2.hs, interpreted )
Ok, modules loaded: Clocks, Fifo2, Test.Fifo2.
*Test.Fifo2> runTest
[False,False,False,False,False,*** Exception: Ix{Int}.index: Index (3) out of range ((0,2))

christiaanb, 2015-06-18:
Thanks for the report, it should be fixed now...

closed 2015-06-18 by christiaanb

masterdezign, 2015-06-18:
Yes, now it works. Thanks