clash-lang/clash-compiler #62: This piece of code eat up all CPU when generating verilog

Alaya-in-Matrix, 2015-06-20:
```haskell
module Learn where 
import CLaSH.Prelude
fib :: Signal (Unsigned 64)
fib = (register 1 fib) + (register 0 $ register 0 fib)
topEntity = fib 
```

I read in the tutorial that `Signal` is represented by infinite list, so I am trying to generate a fib sequence. the code above works well when testing using `clash --interactive`, but when I try to generate verilog code, clash will eat up all CPU and doesn't give error message.

closed 2015-06-20 by christiaanb

christiaanb, 2015-06-20:
Thanks for the report. I've fixed it, but not completely to my satisfaction...

As you might know, the CLaSH compiler uses the GHC compiler as a front-end. In this case, I receive code from GHC that has performed one unrolling of the recursive `fib` function, but spread over multiple functions. The result: the hardware that is generated has 5 registers and 2 adders, instead of the desired 3 registers and 1 adder. The external behaviour remains correct of course.

The following code will ensure that GHC doesn't do any mangling of the recursive functions, and results in a circuit with the 3 registers and the single adder:

```haskell
module Fib where

import CLaSH.Prelude

topEntity :: Signal (Unsigned 64)
topEntity = fib
  where
    {-# NOINLINE fib #-}
    fib =  register 1 fib + register 0 (register 0 fib)
```

Trying to re-roll the recursive step is going to be on my low-priority todo list.... which means I don't think I'll fix it any time soon, given that the generated hardware is correct, just slightly too big.

christiaanb, 2015-06-22:
OK, I lied a bit about being this being low priority. The way the CLaSH compiler was handling recursive function was still bothering me. So I tweaked the way we combine mutually recursive functions to a single recursive functions.

The original example now results in a circuit containing 3 registers and 1 adder.

commit 469f663d (2015-06-20): Do not inline closed binders that are part of a recursive group

Fixes #62