clash-lang/clash-compiler #64: Fail to generate verilog when using "quot" and "div" on Index

Alaya-in-Matrix, 2015-06-24:
Fail to generate verilog code for the below code

```haskell
topEntity :: Index 49 -> Index 49
topEntity x = x `div` 2
```

Besides, I write `topEntity x = x `div` 2` because `Index` is not Instance of `Bits` so I can't apply `shiftR` on it.

christiaanb, 2015-06-24:
Thanks for the report. I intend to make a new release tomorrow (25-06-15) which will include fixes for all the bugs that you have reported so far.

closed 2015-06-24 by christiaanb

christiaanb, 2015-06-24:
Additionally, the `Index` type is basically a (failed) experiment. What I would really like is to have something like MyHDLs [intbv](http://docs.myhdl.org/en/latest/manual/hwtypes.html) type in Haskell, but then with the bounds mentioned in the type.

`Index` will probably be removed in the next version of the clash-prelude, and replaced by something like  MyHDLs `intbv` type.

commit fd5a06f9 (2015-06-24): Fix quot and rem primitives for unsigned types

Fixes #64