clash-lang/clash-compiler #67: clash: unable to load package `primitive-0.6'

LovelyYanki, 2015-07-13:
in ubuntu 12.04 after following installation instruction

clash --interactive FIR.hs
CLaSHi, version 0.5.10 (using clash-lib, version 0.5.9):
http://www.clash-lang.org/   for help
[1 of 1] Compiling FIR ( FIR.hs, interpreted )
clash: unable to load package `primitive-0.6'

christiaanb, 2015-07-13:
Hi, could you give me the output of the following commands:

```
ghc --version
ghc-pkg check
```

christiaanb, 2015-07-13:
And could you give me the output of:

```
clash --interactive -v3 FIR.hs
```

LovelyYanki, 2015-07-13:
martin@ubuntu:~/downloads/mscript$ ghc --version
The Glorious Glasgow Haskell Compilation System, version 7.10.1
martin@ubuntu:~/downloads/mscript$ ghc-pkg check
Warning: haddock-interfaces: /home/martin/.cabal/share/doc/i386-linux-ghc-7.10.1/nats-1/html/nats.haddock doesn't exist or isn't a file
martin@ubuntu:~/downloads/mscript$ 

martin@ubuntu:~/downloads/mscript$ clash --interactive -v3 FIR.hs
CLaSHi, version 0.5.10 (using clash-lib, version 0.5.9):
http://www.clash-lang.org/  :? for help
Glasgow Haskell Compiler, Version 7.10.1, stage 2 booted by GHC version 7.8.4
Using binary package database: /opt/ghc/7.10.1/lib/ghc-7.10.1/package.conf.d/package.cache
Using binary package database: /home/martin/.ghc/i386-linux-7.10.1/package.conf.d/package.cache
wired-in package ghc-prim mapped to ghc-prim-0.4.0.0-361e8e2c9cf389d2a3e7a0e22f6c7443
wired-in package integer-gmp mapped to integer-gmp-1.0.0.0-c95adc98e5a4d89e0ee310717e5e2c43
wired-in package base mapped to base-4.8.0.0-fc683f3bc9a20f2c5e9cfd686d3063fe
wired-in package rts mapped to builtin_rts
wired-in package template-haskell mapped to template-haskell-2.10.0.0-44fef4903d5ebb1b4f10c81ce80bf2dc
wired-in package ghc mapped to ghc-7.10.1-c68fad7c073b01c18711eb22d66d9d8d
wired-in package dph-seq not found.
wired-in package dph-par not found.
Hsc static flags: 
Loading package ghc-prim-0.4.0.0 ... linking ... done.
*** gcc:
/usr/bin/gcc -U__i686 -fno-stack-protector -DTABLES_NEXT_TO_CODE -L/opt/ghc/7.10.1/lib/ghc-7.10.1/integ_2aU3IZNMF9a7mQ0OzsZ0dS --print-file-name libgmp.so
Loading package integer-gmp-1.0.0.0 ... linking ... done.
Loading package base-4.8.0.0 ... linking ... done.
wired-in package ghc-prim mapped to ghc-prim-0.4.0.0-361e8e2c9cf389d2a3e7a0e22f6c7443
wired-in package integer-gmp mapped to integer-gmp-1.0.0.0-c95adc98e5a4d89e0ee310717e5e2c43
wired-in package base mapped to base-4.8.0.0-fc683f3bc9a20f2c5e9cfd686d3063fe
wired-in package rts mapped to builtin_rts
wired-in package template-haskell mapped to template-haskell-2.10.0.0-44fef4903d5ebb1b4f10c81ce80bf2dc
wired-in package ghc mapped to ghc-7.10.1-c68fad7c073b01c18711eb22d66d9d8d
wired-in package dph-seq not found.
wired-in package dph-par not found.
Loading package transformers-0.4.2.0 ... linking ... done.
Loading package array-0.5.1.0 ... linking ... done.
Loading package deepseq-1.4.1.1 ... linking ... done.
Loading package containers-0.5.6.2 ... linking ... done.
Loading package hoopl-3.10.0.2 ... linking ... done.
Loading package bytestring-0.10.6.0 ... linking ... done.
Loading package binary-0.7.3.0 ... linking ... done.
Loading package filepath-1.4.0.0 ... linking ... done.
Loading package time-1.5.0.1 ... linking ... done.
*** gcc:
/usr/bin/gcc -U__i686 -fno-stack-protector -DTABLES_NEXT_TO_CODE -L/opt/ghc/7.10.1/lib/ghc-7.10.1/unix_G4Yo1pNtYrk8nCq1cx8P9d --print-file-name librt.so
*** gcc:
/usr/bin/gcc -U__i686 -fno-stack-protector -DTABLES_NEXT_TO_CODE -L/opt/ghc/7.10.1/lib/ghc-7.10.1/unix_G4Yo1pNtYrk8nCq1cx8P9d --print-file-name libutil.so
*** gcc:
/usr/bin/gcc -U__i686 -fno-stack-protector -DTABLES_NEXT_TO_CODE -L/opt/ghc/7.10.1/lib/ghc-7.10.1/unix_G4Yo1pNtYrk8nCq1cx8P9d --print-file-name libdl.so
*** gcc:
/usr/bin/gcc -U__i686 -fno-stack-protector -DTABLES_NEXT_TO_CODE -L/opt/ghc/7.10.1/lib/ghc-7.10.1/unix_G4Yo1pNtYrk8nCq1cx8P9d --print-file-name libpthread.so
Loading package unix-2.7.1.0 ... linking ... done.
Loading package directory-1.2.2.0 ... linking ... done.
Loading package bin-package-db-0.0.0.0 ... linking ... done.
Loading package pretty-1.1.2.0 ... linking ... done.
Loading package process-1.2.3.0 ... linking ... done.
Loading package hpc-0.6.0.2 ... linking ... done.
Loading package template-haskell-2.10.0.0 ... linking ... done.
Loading package ghc-7.10.1 ... linking ... done.
Loading package ghc-tcplugins-extra-0.1 ... linking ... done.
Loading package ghc-typelits-natnormalise-0.3 ... linking ... done.
Coercing a value in getValueSafely...
Successfully evaluated coercion
*** Chasing dependencies:
Chasing modules from: 
Stable obj: []
Stable BCO: []
unload: retaining objs []
unload: retaining bcos []
Ready for upsweep []
Upsweep completely successful.
*** Deleting temp files:
Deleting: 
*** Chasing dependencies:
Chasing modules from: *FIR.hs
Stable obj: []
Stable BCO: []
unload: retaining objs []
unload: retaining bcos []
Ready for upsweep
  [NONREC
      ModSummary {
         ms_hs_date = 2015-07-11 00:59:01.367425486 UTC
         ms_mod = FIR,
         ms_textual_imps = [import CLaSH.Prelude]
         ms_srcimps = []
      }]
*** Deleting temp files:
Deleting: 
compile: input file FIR.hs
Created temporary directory: /tmp/ghc3712_0
*** Checking old interface for FIR:
[1 of 1] Compiling FIR              ( FIR.hs, interpreted )
*** Parser:
*** Renamer/typechecker:
Coercing a value in getValueSafely...
Successfully evaluated coercion
*** Simplify:
*** CorePrep:
*** ByteCodeGen:
Loading package primitive-0.6 ... linking ... clash: unable to load package `primitive-0.6'

christiaanb, 2015-07-13:
And the output of?:

```
ghc-pkg list
ghci Fir.hs
```

LovelyYanki, 2015-07-15:
martin@ubuntu:~/Downloads$ ghc-pkg list
/opt/ghc/7.10.1/lib/ghc-7.10.1/package.conf.d
   Cabal-1.22.2.0
   array-0.5.1.0
   base-4.8.0.0
   bin-package-db-0.0.0.0
   binary-0.7.3.0
   bytestring-0.10.6.0
   containers-0.5.6.2
   deepseq-1.4.1.1
   directory-1.2.2.0
   filepath-1.4.0.0
   ghc-7.10.1
   ghc-prim-0.4.0.0
   haskeline-0.7.2.1
   hoopl-3.10.0.2
   hpc-0.6.0.2
   integer-gmp-1.0.0.0
   pretty-1.1.2.0
   process-1.2.3.0
   rts-1.0
   template-haskell-2.10.0.0
   terminfo-0.4.0.1
   time-1.5.0.1
   transformers-0.4.2.0
   unix-2.7.1.0
   xhtml-3000.2.1
/home/martin/.ghc/i386-linux-7.10.1/package.conf.d
   ListLike-4.2.0
   QuickCheck-2.8.1
   StateVar-1.1.0.0
   adjunctions-4.2.1
   aeson-0.9.0.1
   attoparsec-0.13.0.1
   base-orphans-0.4.0
   bifunctors-5
   clash-lib-0.5.9
   clash-prelude-0.9.1
   clash-systemverilog-0.5.7
   clash-verilog-0.5.7
   clash-vhdl-0.5.8
   comonad-4.2.6
   concurrent-supply-0.1.7.1
   contravariant-1.3.1.1
   data-default-0.5.3
   data-default-class-0.0.1
   data-default-instances-base-0.0.1
   data-default-instances-containers-0.0.1
   data-default-instances-dlist-0.0.1
   data-default-instances-old-locale-0.0.1
   distributive-0.4.4
   dlist-0.7.1.1
   errors-2.0.0
   exceptions-0.8.0.2
   fgl-5.5.1.0
   fmlist-0.9
   free-4.12.1
   ghc-tcplugins-extra-0.1
   ghc-typelits-natnormalise-0.3
   hashable-1.2.3.3
   kan-extensions-4.2.2
   lens-4.12
   mtl-2.2.1
   nats-1
   old-locale-1.0.0.7
   parallel-3.2.0.6
   prelude-extras-0.4
   primitive-0.6
   profunctors-5.1.1
   random-1.1
[... 159 lines cut ...]
unload: retaining objs []
unload: retaining bcos []
Ready for upsweep
  [NONREC
      ModSummary {
         ms_hs_date = 2015-07-10 11:57:31.718435769 UTC
         ms_mod = FIR,
         ms_textual_imps = [import CLaSH.Prelude]
         ms_srcimps = []
      }]
*** Deleting temp files:
Deleting: 
compile: input file FIR.hs
Created temporary directory: /tmp/ghc2902_0
*** Checking old interface for FIR:
[1 of 1] Compiling FIR              ( FIR.hs, interpreted )
*** Parser:
*** Renamer/typechecker:
Coercing a value in getValueSafely...
Successfully evaluated coercion
*** Simplify:
*** CorePrep:
*** ByteCodeGen:
Loading package random-1.1 ... linking ... done.
Loading package primitive-0.6 ... linking ... clash: unable to load package `primitive-0.6'

christiaanb, 2015-07-15:
Sorry, can you give me the output of one more:

```
ghci -XDataKinds -XNoImplicitPrelude -XTypeOperators -XTemplateHaskell -v3 FIR.hs
```

The thing is.... I cannot reproduce the bug on any of my machines.

LovelyYanki, 2015-07-15:
martin@ubuntu:~/Downloads$ ghci -XDataKinds -XNoImplicitPrelude -XTypeOperators -XTemplateHaskell -v3 FIR.hs
GHCi, version 7.10.1: http://www.haskell.org/ghc/  :? for help
Glasgow Haskell Compiler, Version 7.10.1, stage 2 booted by GHC version 7.8.4
Using binary package database: /opt/ghc/7.10.1/lib/ghc-7.10.1/package.conf.d/package.cache
Using binary package database: /home/martin/.ghc/i386-linux-7.10.1/package.conf.d/package.cache
wired-in package ghc-prim mapped to ghc-prim-0.4.0.0-361e8e2c9cf389d2a3e7a0e22f6c7443
wired-in package integer-gmp mapped to integer-gmp-1.0.0.0-c95adc98e5a4d89e0ee310717e5e2c43
wired-in package base mapped to base-4.8.0.0-fc683f3bc9a20f2c5e9cfd686d3063fe
wired-in package rts mapped to builtin_rts
wired-in package template-haskell mapped to template-haskell-2.10.0.0-44fef4903d5ebb1b4f10c81ce80bf2dc
wired-in package ghc mapped to ghc-7.10.1-c68fad7c073b01c18711eb22d66d9d8d
wired-in package dph-seq not found.
wired-in package dph-par not found.
Hsc static flags: 
Loading package ghc-prim-0.4.0.0 ... linking ... done.
*** gcc:
/usr/bin/gcc -U__i686 -fno-stack-protector -DTABLES_NEXT_TO_CODE -L/opt/ghc/7.10.1/lib/ghc-7.10.1/integ_2aU3IZNMF9a7mQ0OzsZ0dS --print-file-name libgmp.so
Loading package integer-gmp-1.0.0.0 ... linking ... done.
Loading package base-4.8.0.0 ... linking ... done.
wired-in package ghc-prim mapped to ghc-prim-0.4.0.0-361e8e2c9cf389d2a3e7a0e22f6c7443
wired-in package integer-gmp mapped to integer-gmp-1.0.0.0-c95adc98e5a4d89e0ee310717e5e2c43
wired-in package base mapped to base-4.8.0.0-fc683f3bc9a20f2c5e9cfd686d3063fe
wired-in package rts mapped to builtin_rts
wired-in package template-haskell mapped to template-haskell-2.10.0.0-44fef4903d5ebb1b4f10c81ce80bf2dc
wired-in package ghc mapped to ghc-7.10.1-c68fad7c073b01c18711eb22d66d9d8d
wired-in package dph-seq not found.
wired-in package dph-par not found.
*** Chasing dependencies:
Chasing modules from: 
Stable obj: []
Stable BCO: []
unload: retaining objs []
unload: retaining bcos []
Ready for upsweep []
Upsweep completely successful.
*** Deleting temp files:
Deleting: 
*** Chasing dependencies:
Chasing modules from: *FIR.hs
Stable obj: []
Stable BCO: []
unload: retaining objs []
unload: retaining bcos []
Ready for upsweep
  [NONREC
      ModSummary {
         ms_hs_date = 2015-07-10 11:57:31.718435769 UTC
         ms_mod = FIR,
         ms_textual_imps = [import CLaSH.Prelude]
         ms_srcimps = []
      }]
*** Deleting temp files:
Deleting: 
compile: input file FIR.hs
Created temporary directory: /tmp/ghc3489_0
*** Checking old interface for FIR:
[1 of 1] Compiling FIR              ( FIR.hs, interpreted )
*** Parser:
*** Renamer/typechecker:
*** Simplify:
*** CorePrep:
*** ByteCodeGen:
Loading package transformers-0.4.2.0 ... linking ... done.
Loading package array-0.5.1.0 ... linking ... done.
Loading package deepseq-1.4.1.1 ... linking ... done.
Loading package bytestring-0.10.6.0 ... linking ... done.
Loading package containers-0.5.6.2 ... linking ... done.
Loading package binary-0.7.3.0 ... linking ... done.
Loading package pretty-1.1.2.0 ... linking ... done.
Loading package time-1.5.0.1 ... linking ... done.
[... 68 lines cut ...]
*** CorePrep:
*** ByteCodeGen:
*** Simplify:
*** CorePrep:
*** ByteCodeGen:
*** Desugar:
Result size of Desugar (after optimization)
  = {terms: 150, types: 238, coercions: 111}
*** Simplifier:
Result size of Simplifier iteration=1
  = {terms: 160, types: 279, coercions: 296}
Result size of Simplifier
  = {terms: 150, types: 269, coercions: 255}
*** Tidy Core:
Result size of Tidy Core = {terms: 150, types: 269, coercions: 255}
*** CorePrep:
Result size of CorePrep = {terms: 258, types: 382, coercions: 255}
*** ByteCodeGen:
Upsweep completely successful.
*** Deleting temp files:
Deleting: /tmp/ghc3489_0/ghc3489_2.c /tmp/ghc3489_0/ghc3489_1.o
Warning: deleting non-existent /tmp/ghc3489_0/ghc3489_2.c
Warning: deleting non-existent /tmp/ghc3489_0/ghc3489_1.o
Ok, modules loaded: FIR.
*FIR>

christiaanb, 2015-07-15:
Thanks for all the help. I think I know what might be going on.
Can you try the following:

First:
```
clash -- --info | grep "GHC Dynamic"
```

If the output of the above is:
```
 ,("GHC Dynamic","NO")
```

Can you then try:
```
cabal install clash-ghc --enable-executable-dynamic
clash --interactive FIR.hs
```

And if the output of `clash -- --info | grep "GHC Dynamic"` is:
```
,("GHC Dynamic","YES")
```

Can you then try:
```
cabal install clash-ghc --disable-executable-dynamic
clash --interactive FIR.hs
```

LovelyYanki, 2015-07-15:
run this with no error output
cabal install clash-ghc --enable-executable-dynamic
clash --interactive FIR.hs

when run this disable, it has primitive 0.6 error
cabal install clash-ghc --disable-executable-dynamic
clash --interactive FIR.hs

christiaanb, 2015-07-15:
Thank you for all the help. I will update the installation instructions so that people install clash using `cabal install clash-ghc --enable-executable-dynamic`

closed 2015-07-15 by christiaanb