clash-lang/clash-compiler #68: Cabal file does not specify version upper bounds

thomie, 2015-08-01:
To reproduce an [issue](https://ghc.haskell.org/trac/ghc/ticket/10134) reported to the ghc bug tracker, I tried running the following command in a cabal sandbox:
$ cabal install clash-ghc==0.4.1 --with-ghc=ghc-7.8.4

After it installed all the dependencies, I get the following error:
```
src-ghc/CLaSH/GHC/GHC2Core.hs:138:35:
     Couldn't match type ‘Unbound.Name a32’
                   with ‘unbound-generics-0.1.2.1:Unbound.Generics.LocallyNameless.Name.Name
                           C.TyCon’
     NB: ‘Unbound.Name’
           is defined in ‘Unbound.LocallyNameless.Name’
               in package ‘unbound-0.4.3.1’
         ‘unbound-generics-0.1.2.1:Unbound.Generics.LocallyNameless.Name.Name’
           is defined in ‘Unbound.Generics.LocallyNameless.Name’
               in package ‘unbound-generics-0.1.2.1’
```

I'm using cabal constraints from this file: https://www.stackage.org/lts-1.15/cabal.config?global=true

The problem seems to be:
* clash-ghc-0.4.1 depends on clash-lib (>=0.4.1)
* cabal picks clash-lib-0.5
* clash-ghc-0.4.1 (which depends on unbound (>=0.4.0.2)) is not compatible with clash-lib-0.5 (which depends on unbound-generics (>=0.1))

Specifying the following upper bound fixed it for me: clash-lib < 0.5.

Even if older versions of clash-ghc are no longer supported, it would be nice to be able to still install them. Please consider adding version upperbounds to the clash-ghc.cabal file.

christiaanb, 2015-08-01:
Should I just update the `.cabal` files on hackage?

thomie, 2015-08-01:
Yes, I think that would work. But I'd also set upper bounds in the .cabal file in this repository, to prevent such issues in the future.

Another problem: `cabal install --with-ghc=ghc-7.8.4 clash-ghc` installs `clash-0.5`. But that version is I think unusable, because although it specifies `ghc (>=7.8)`, I get the following error:
```
Can't parse "/opt/ghc/7.10.2/lib/ghc-7.10.2/platformConstants"`.
```

So `clash-0.5` should I think depend on `ghc (=7.10)`.

christiaanb, 2015-08-02:
I've fixed the relevant upper bounds, and just marked `0.5` as deprecated. I know the `0.4` releases compile against ghc 7.8, and `0.5.1` needs ghc 7.10. However, `0.5` should be able to compile against ghc 7.8... I don't know what's going on, so I just marked it as deprecated on hackage.

closed 2015-08-02 by thomie