clash-lang/clash-compiler #69: Bug in VHDL ROM generation

polygonhell, 2015-08-13:
I'm creating a synchronous rom as a part of a project exporting as VHDL I get the following as a part of the output.

Note the lines

~~~
ROM <= (array_of_unsigned_8'(0 to (512)-1 => unsigned'("00000011")));
~~~
and 
~~~
rom : process (system1000)
~~~

Since VHDL is case insensitive this causes a compile error at least with the xilinx tools.
In case this isn't a trivial repro case I've checked the code that creates the issue into https://github.com/polygonhell/Clash6502 I'll leave it in the broken state for a week or so but you want commit bcb3d4655a165aef9b91e2d60a5600cc4c34eb4e

You likely won't get the makefile to work, but you can generate the VHDL with 

clash --vhdl Main.hs

~~~
  -- rom begin
  rom_Main_topEntityzm_tup_3_n_14 : block
    signal ROM  : array_of_unsigned_8(0 to 511);
    signal rd   : integer range 0 to 512-1;
    signal dout : unsigned(7 downto 0);
  begin
    ROM <= (array_of_unsigned_8'(0 to (512)-1 => unsigned'("00000011")));
  
    rd <= bodyVar_2
    -- pragma translate_off
          mod 512
    -- pragma translate_on
          ;
  
    rom : process (system1000)
    begin
      if (rising_edge(system1000)) then
        dout <= ROM(rd);
      end if;
    end process;
  
    tmp_10 <= dout;
  end block;
~~~

christiaanb, 2015-08-13:
Ah, I see how this bug slipped by... I only included the file-based ROM primitives in the test-suite, in which the process is called `romFile`. I'll post a fix shortly, as a work-around you can put the following `CLaSH.Prelude.ROM.json` file in your working directory:

```json
[ { "BlackBox" :
    { "name" : "CLaSH.Prelude.ROM.rom#"
    , "type" :
"rom# :: KnownNat n      -- ARG[0]
      => SClock clk      -- clk,  ARG[1]
      -> Vec n a         -- init, ARG[2]
      -> Signal' clk Int -- rd,   ARG[3]
      -> Signal' clk a"
    , "templateD" :
"-- rom begin
rom_~COMPNAME_~SYM[0] : block
  signal ROM  : ~TYP[2];
  signal rd   : integer range 0 to ~LIT[0]-1;
  signal dout : ~TYPO;
begin
  ROM <= ~ARG[2];

  rd <= ~ARG[3]
  -- pragma translate_off
        mod ~LIT[0]
  -- pragma translate_on
        ;

  romSync : process (~CLK[1])
  begin
    if (rising_edge(~CLK[1])) then
      dout <= ROM(rd);
    end if;
  end process;

  ~RESULT <= dout;
end block;
-- rom end"
    }
  }
]
```

closed 2015-08-13 by christiaanb

christiaanb, 2015-08-13:
Is the work-around working for you? Is there any rush to release a new version?

polygonhell, 2015-08-13:
Thanks for the quick response Christian.
I'm in no rush for a fix, I ended up using BlockRam for this application, I just ran across this when I was experimenting with the various RAM/ROM primitives to see what the Xilinx tools actually did with them.

commit c3fd2668 (2015-08-13): Rename ROM process name

So that it does not conflict with the ROM signal.

Fixes #69