clash-lang/clash-compiler #70: Clash running out of memory on Simple-ish project

polygonhell, 2015-08-23:
The source code is here - https://github.com/polygonhell/Clash6502 
The implementation is incomplete, but trying to generate VHDL code results in Clash using over 40GB's of memory before finally throwing an out of memory.
I assume this is a dupe of #54, but my largest case statement is only 18 way, so I'm filing this as a different issue in case there is something else at play.

christiaanb, 2015-08-23:
I'm on holiday until the 31st of august, so don't have the time to investigate until then. 

Could you try adding `{-# NOINLINE ... #-}` pragma's to the larger functions, such as `Alu.execWithData`, `Alu.addrMode`, `aluOp`, etc. and see if that helps.
If it helps, try removing the pragma's again one-by-one, and see which is the offending one.

Thanks.

polygonhell, 2015-08-23:
No rush on a solution to this, just wanted to make sure you were aware of it.
Setting {-# NOINLINE ... #-} on pretty much anything of any size didn't resolve the issue, or at least it takes more than 30 minutes to run,  it did however greatly reduce the rate of memory growth, when I finally killed it it was using ~12GB of memory.

Looking at the debug output it looks like Alu,addrMode is normalized, but Alu.aluOp never completes normalization, there are about 30K transforms in the log for it, the majority inlineClosed, the last line being ANF, which I assume is rewriting to some normal form.

addrMode is a 16 way "case" and produces ~150 transforms to normalize and aluOp is a 28 way "case" producing >30K transforms so it looks like some sort of exponential complexity during the rewrite of case statements.

At this point I'm assuming the issue is the same as #54

christiaanb, 2015-08-24:
Thanks for the thorough analysis. It confirms two of my suspicions:

1. I should turn off, or at least significantly reduce, the inlining done by the GHC optimiser that I use as part of the CLaSH frontend.
2. CLaSH deals poorly with larger case-statements involving integer/number alternatives.

christiaanb, 2015-08-31:
As a work around, the following `aluOp` compiles much faster:

```haskell
aluOp :: Unsigned 2 -> Unsigned 3 -> Unsigned 3 -> AluOp
aluOp dm addrBits opBits = case dm of
  1 -> case opBits of
         0 -> ORA
         1 -> AND
         2 -> EOR
         3 -> ADC
         4 -> case addrBits of
                2 -> BIT
                _ -> STA
         5 -> LDA
         6 -> CMP
         7 -> SBC
         _ -> ILLEGAL
  2 -> case addrBits of
         0 -> ILLEGAL
         _ -> case opBits of
                0 -> ASL
                1 -> ROL
                2 -> LSR
                3 -> ROR
                4 -> case addrBits of
                       2 -> TXA
                       7 -> ILLEGAL
                       _ -> STX
                5 -> case addrBits of
                       2 -> TAX
                       _ -> LDX
                6 -> DEC
                7 -> INC
                _ -> ILLEGAL
  _ -> ILLEGAL
```

The reason being that I get a much nicer `case`-statement from GHC. 

For your original `aluOp`, GHC gives me code that might work very nicely for execution on a CPU, but leads to the observed code-explosion when converting to a normal form that works well for creating hardware. Sadly, I don't have much influence on how GHC creates these internal case-statements.

christiaanb, 2015-08-31:
Ah, more progress, sort-of:

* Your original `aluOp` results in a local function called `fail`, so what CLaSH gets to see is this:

  ```haskell
aluOp =
λ(ds :: CLaSH.Sized.Internal.Unsigned.Unsigned 2) ->
λ(ds1 :: CLaSH.Sized.Internal.Unsigned.Unsigned 3) ->
λ(ds2 :: CLaSH.Sized.Internal.Unsigned.Unsigned 3) ->
letrec
  fail :: GHC.Prim.Void# -> ALU2.AluOp
  = λ(ds3 :: GHC.Prim.Void#) ->
    case CLaSH.Sized.Internal.Unsigned.eq#
         @2
           ds
           ALU2.aluOp191912956404 of
      GHC.Types.False -> ALU2.ILLEGAL
      GHC.Types.True ->
        letrec
          fail1 :: GHC.Prim.Void# -> ALU2.AluOp
          = λ(ds4 :: GHC.Prim.Void#) ->
            case CLaSH.Sized.Internal.Unsigned.eq#
                 @3
                   ds2
                   ALU2.aluOp171912956402 of
              GHC.Types.False ->
                letrec
                  fail2 :: GHC.Prim.Void# -> ALU2.AluOp
                  = λ(ds5 :: GHC.Prim.Void#) ->
                    case CLaSH.Sized.Internal.Unsigned.eq#
                         @3
                           ds2
                           ALU2.aluOp161912956401 of
                      GHC.Types.False ->
                        letrec
                          fail3 :: GHC.Prim.Void# -> ALU2.AluOp
                          = λ(ds6 :: GHC.Prim.Void#) ->
                            case CLaSH.Sized.Internal.Unsigned.eq#
                                 @3
                                   ds2
                                   ALU2.aluOp151912956400 of
                              GHC.Types.False ->
                                letrec
                                  fail4 :: GHC.Prim.Void# -> ALU2.AluOp
                                  = λ(ds7 :: GHC.Prim.Void#) ->
                                    case CLaSH.Sized.Internal.Unsigned.eq#
                                         @3
                                           ds2
                                           ALU2.aluOp141912956399 of
                                      GHC.Types.False ->
                                        letrec
                                          fail5 :: GHC.Prim.Void# -> ALU2.AluOp
                                          = λ(ds8 :: GHC.Prim.Void#) ->
                                            letrec
                                              fail6 :: GHC.Prim.Void# -> ALU2.AluOp
                                              = λ(ds9 :: GHC.Prim.Void#) ->
                                                letrec
                                                  fail7 :: GHC.Prim.Void# -> ALU2.AluOp
                                                  = λ(ds10 :: GHC.Prim.Void#) ->
                                                    case CLaSH.Sized.Internal.Unsigned.eq#
                                                         @3
                                                           ds2
                                                           ALU2.aluOp121912956398 of
                                                      GHC.Types.False ->
                                                        letrec
                                                          fail8 :: GHC.Prim.Void# -> ALU2.AluOp
                                                          = λ(ds11 :: GHC.Prim.Void#) ->
                                                            case CLaSH.Sized.Internal.Unsigned.eq#
                                                                 @3
                                                                   ds2
[... 244 lines cut ...]
                              fail1
                                GHC.Prim.void#
                            GHC.Types.True ->
                              case CLaSH.Sized.Internal.Unsigned.eq#
                                   @3
                                     ds2
                                     ALU2.aluOp121912956398 of
                                GHC.Types.False ->
                                  fail1
                                    GHC.Prim.void#
                                GHC.Types.True -> ALU2.BIT
                     GHC.Types.True -> ALU2.ADC
                 GHC.Types.True -> ALU2.EOR
             GHC.Types.True -> ALU2.AND
         GHC.Types.True -> ALU2.ORA
```
* All those `letrec`s are local, _non-recursive_, functions.
* _Non-recursive_ local functions normally get _inlined_ because we try to preserve as much as the first-order function hierarchy in the generated hardware component hierarchy. Basically what this means is that we prefer if every top-level function in your Haskell/CLaSH code becomes an entity/module in your VHDL/Verilog code.
* Another option is to turn local functions into global functions, this is something we always do with local functions when they are _recursive_. 

The _downside_ to turning all local functions into global functions is that, for example in your `aluOp` function, that you will get __11__ `fail` components in your VHDL/Verilog, each with their own VHDL/Verilog file. So you will get components with strange components, e.g. `fail`, which never existed in your Haskell/CLaSH code, suddenly turn up in your VHDL/Verilog code.

The _upside_ to turning all local functions into global functions is, as I just tested, that it will significantly reduce the compile time and memory usage as there isn't this exponential behaviour which CLaSH seems to exhibit for very large expressions.

So yeah... I'm not completely sure yet what to do. Of course, running out of memory is clearly a (performance) bug, but I really want to preserve the original function hierarchy, and not introduce too many components which never existed as Haskell-level functions.

polygonhell, 2015-08-31:
Yeah that's pretty ugly.
But I guess it's the nature of the limitations placed on the Core intermediate form specifically that expressions can only be applied to atoms, so much of the logic gets lifted into the fail functions. So I don't think the compilers going to stop doing this.

I assume the exponential behavior is the result of recursively inlining the nested fail functions, in which case there likely isn't a good way to avoid the exponential cost without not doing that.

I agree it's undesirable to have functions show up in the VHDL that weren't in the original code, I guess the only real solution is to not inline and to mark functions like this in someway and move their implementation into the file associated with the actual source function. 
But I would also guess that this is a none trivial change.

BTW thanks for the work around.

commit c57ef45b (2015-09-03): Record provenance in lifted binders

For now, we just prefix the name of the function from which
a lifted binder originated, to the name of the new global
binder. At least this reflects the provenance of certain
"weird" components which sometimes appear because of the
way GHC compiles pattern matching functions to case-statements;
such as what happens in issue #70.

Fixes #70
closed 2015-09-03 by christiaanb