clash-lang/clash-compiler #78: CLaSH fails to generate VHDL when map is applied

masterdezign, 2015-09-09:
The following module compiles, but fails to generate VHDL with an error "*** Exception: CLaSH.Netlist(269): Unknown function: mem_165".
```
module TestMap
  where

import CLaSH.Prelude
import CLaSH.Prelude.Explicit

type System50 = Clk "system" 72
system50 :: SClock System50
system50 = sclock
type Sig1 = Signal' System50
regEn1 = regEn' system50
bundle1 = bundle' system50

type Word1 = SFixed 9 9
type RomFunction = (Unsigned 9 -> Word1)

coref :: RomFunction -> (Sig1 Word1, Sig1 Bool) -> Sig1 Word1
coref romf (bk, _) = (*) <$> bk <*> (signal $ romf 0)

mf ::
  (KnownNat n, KnownNat n1, KnownNat n2)
  => (Vec n RomFunction, SNat n1, SNat n2)
  -> (Sig1 Word1, Sig1 Bool)
  -> Sig1 Word1
mf (romFunctions, nI, coefN) (bk, dPulse) = (!!) <$> (bundle1 results) <*> curCore
  where

    g (mem, coreIndex) = initCore (mem, curCore .==. coreIndex)
    results = map g $ zip romFunctions (iterateI (+1) 0)

    initCore (romF, readSel) = (coref romF) (bk, readSel)

    read = dPulse

    cores = length results
    curCore :: Sig1 (Unsigned 1)
    curCore = regEn1 0 read (curCore + 1)

romF1 :: RomFunction
romF1 addr = 0
romF2 :: RomFunction
romF2 addr = 1

topEntity = mf ((romF1 :> romF2 :> Nil), d2, d8)
```

However, the equivalent, but less generic version generates VHDL perfectly fine:
```
mf (romFunctions, nI, coefN) (bk, dPulse) = (!!) <$> (bundle1 results) <*> curCore
  where

    results = (initCore (romF1, curCore .==. 0))
      :> (initCore (romF2, curCore .==. 1))
      :> Nil

    initCore (romF, readSel) = (coref romF) (bk, readSel)

    read = dPulse

    cores = length results
    curCore :: Sig1 (Unsigned 1)
    curCore = regEn1 0 read (curCore + 1)
```

The only difference is:
```
28,29c28,30
<     g (mem, coreIndex) = initCore (mem, curCore .==. coreIndex)
<     results = map g $ zip romFunctions (iterateI (+1) 0)
---
>     results = (initCore (romF1, curCore .==. 0))
>       :> (initCore (romF2, curCore .==. 1))
>       :> Nil
```

christiaanb, 2015-09-10:
Hi, sorry for the unclear error message. 

It's due to the fact that the `zip` function is implemented using the `zipWith` function, which is a higher-order primitive in CLaSH. At the moment, these higher-order primitives do not handle vectors containing functions, so this bug is a duplicate of #25.

Sorry for the inconvenience. I hope to fix #25 and this bug as part of a major overhaul of the CLaSH compiler internals, but that is going to take some time.

masterdezign, 2015-09-10:
Hello Christiaan,

a version _without_ zip fails as well:

```
mf (romFunctions, nI, coefN) (bk, dPulse) = (!!) <$> (bundle1 results) <*> curCore
  where

    g (mem, coreIndex) = initCore (mem, curCore .==. coreIndex)
    results = map g $ (romF1, 0) :> (romF2, 1) :> Nil

    initCore (romF, readSel) = (coref romF) (bk, readSel)

    read = dPulse

    cores = length results
    curCore :: Sig1 (Unsigned 1)
    curCore = regEn1 0 read (curCore + 1)
```

It produces ```*** Exception: CLaSH.Netlist(269): Unknown function: mem_146``` when generating VHDL.

However, it would be _extremely_ handy to be able to use ```map```.

Thank you

christiaanb, 2015-09-10:
I'll see if I can hack around it, I've actually wanted to be able to translate e.g. `zipWith ($) xs ys` for a long time.

closed 2015-09-11 by christiaanb

christiaanb, 2015-09-11:
OK, i've hacked around it. See https://github.com/clash-lang/clash-compiler/tree/master/tests/shouldwork/HOPrim for some examples that work now.

I'll make a new release containing this new feature after the weekend. Or perhaps during the weekend if I'm bored ;-)

masterdezign, 2015-09-11:
Thank you a lot for this update :+1:

christiaanb, 2015-09-14:
This fix is included in the, just released, 0.5.14 version of `clash-ghc`

commit f6673fd5 (2015-09-11): Unroll HO primitives when they would lead to non-representable let-bindings

Fixes #78
Fixes #25