clash-lang/clash-compiler #80: Is there any way to get CLaSH to work with bounded polymorphic recursion?

wyager, 2015-09-29:
With the following code, 

```
{-# LANGUAGE GADTs #-}

import CLaSH.Prelude

data Tree n a where
    Leaf :: a -> Tree 0 a
    Node :: Tree n a -> Tree n a -> Tree (n+1) a

tfold :: (a -> a -> a) -> Tree n a -> a
tfold f (Leaf a) = a
tfold f (Node l r) = f (tfold f l) (tfold f r)

topEntity :: Tree 2 Int -> Int
topEntity = tfold (+)

tree :: Tree 2 Int
tree = Node (Node (Leaf 0) (Leaf 5)) (Node (Leaf 1) (Leaf 2))

main = print $ tfold (+) tree
```

The CLaSH compiler (understandably) emits

```
CLaSH.Normalize(109): Expr belonging to bndr: Main.tfold2 (:: ∀ (n :: GHC.TypeLits.Nat).
(GHC.Types.Int -> GHC.Types.Int -> GHC.Types.Int)
-> Main.Tree n GHC.Types.Int -> GHC.Types.Int) remains recursive after normalization:
```

Is there any way to make the CLaSH compiler realize that the type is actually different? (i.e. keep track of the value of `n` instead of just universally quantifying it). Then it would realize that it's not actually generating the same circuit each time. 

On a related note, how does this currently work with sized vectors? Are there no (polymorphically) recursive function definitions?

Ericson2314, 2015-09-29:
Currently, CLaSH cannot compile such recursion. This is an known annoyance, and will be fixed when the current core language is replaced with a more expressive one---indeed this is one of the major motivations for that overhaul. Christiaan's thesis explains how the new system will work.

Currently the recursive vector functions are not complied, but instead are replaced with a a Verilog/VHDL black box template (this is analogous to inline assembly conditionally used only when outputting Verilog/VHDL).

wyager, 2015-09-29:
Great, that's what I wanted to know. Closing the issue, since it's on the table already. Cheers!

closed 2015-09-29 by wyager

christiaanb, 2015-09-30:
Incidentally, the Vector [fold](http://hackage.haskell.org/package/clash-prelude-0.9.3/docs/CLaSH-Sized-Vector.html#v:fold) function will become synthesisable in the next minor release (due this week). 
[fold](http://hackage.haskell.org/package/clash-prelude-0.9.3/docs/CLaSH-Sized-Vector.html#v:fold) reduces a vector in a tree like structure.

This wasn't synthesisable in earlier versions of CLaSH because it needs a hybrid of built-in function-unrolling knowledge specifically for [fold](http://hackage.haskell.org/package/clash-prelude-0.9.3/docs/CLaSH-Sized-Vector.html#v:fold), and a black box (inline assembly) specifically when we are reducing a vector that has a power of two length.

If you're interested into the exact details why unrolling recursive definitions working on GADTs is not trivial, and hence not implemented yet, you can reed section 4.5.1 of my [thesis](http://doc.utwente.nl/93962/).