clash-lang/clash-compiler #81: Parser bug?

wyager, 2015-10-04:
I'm getting some strange behavior. Compile this:

```
{-# LANGUAGE GeneralizedNewtypeDeriving #-}

module Defs (S,Addr(..),W(..),PC(..)) where

import CLaSH.Prelude 

type S a = Signal a

newtype Addr = Addr (addr :: BitVector 16) deriving (Eq, Num, Show)
newtype PC   = PC   (pc   :: BitVector 16) deriving (Eq, Num, Show)
newtype W    = W    (w    :: BitVector 16) deriving (Eq, Num, Show)
```

And you get ```Defs.hs:9:40: parse error on input ‘16’```.

Remove the ```addr ::```, ```pc ::```, etc. and it works fine.

christiaanb, 2015-10-04:
Ah, the syntax for field names is:

```haskell
newtype Addr = Addr {addr :: BitVector 16} deriving (Eq, Num, Show)
newtype PC   = PC   {pc   :: BitVector 16} deriving (Eq, Num, Show)
newtype W    = W    {w    :: BitVector 16} deriving (Eq, Num, Show)
```

So curly braces instead of parenthesis.

wyager, 2015-10-04:
Hah, I can't believe I made that mistake. I even opened another hs file, re-typed one of the definitions (correctly this time), and compiled it with ghc to test. Oops! Thanks.

closed 2015-10-04 by wyager