clash-lang/clash-compiler #82: clash won't run when not compiled with usual ghc

ggreif, 2015-10-06:
I managed to build `clash` with a development version of ghc (`ghc-7.11.20150407` happened to be lying around) involving minimal changes. Alas, on starting it I got
```
$ /home/ggreif/.cabal/bin/clash --interactive
Can't parse "/home/ggreif/lib/ghc-7.11.20150928/platformConstants"
```
(Please note the differing versions.) The reason is that
``` haskell
ghcLibDir :: IO FilePath
ghcLibDir = do (libDir,exitCode) <- getProcessOutput "ghc --print-libdir"
    ...
```
queries the _canonical_ `ghc` that is installed, which may have a different libdir than the `ghc` that built `clash`. Then the `platformConstants` file might be incompatible. The problem will also manifest itself when somebody installs a newer `ghc` without rebuilding `clash` itself.

I suggest doing
``` haskell
ghcLibDir :: IO FilePath
ghcLibDir = do (libDir,exitCode) <- getProcessOutput "ghc-" ++ TOOL_VERSION_ghc ++ " --print-libdir"
```
to access the specific version of `ghc` that was used to build `clash`.

Then I discovered that the functions `ghcLibDir` and `getProcessOutput` are duplicated:
```
$ git grep getProcessOutput
clash-ghc/src-bin/Main.hs:getProcessOutput :: String -> IO (String, ExitCode)
clash-ghc/src-bin/Main.hs:getProcessOutput command =
clash-ghc/src-ghc/CLaSH/GHC/LoadModules.hs:getProcessOutput :: String -> IO (String, ExitCode)
clash-ghc/src-ghc/CLaSH/GHC/LoadModules.hs:getProcessOutput command =
```
This could be refactored. Any suggestion which one to keep? Should I create a new common module?

christiaanb, 2015-10-06:
I prefer a new common module in the `src-ghc` dir. Perhaps `CLaSH.GHC.Util.`?

ggreif, 2015-10-06:
no need. I can export
```
module CLaSH.GHC.LoadModules
  ( loadModules
  , ghcLibDir
  )
```
and use from Main.hs. Easy.

christiaanb, 2015-10-06:
Fixed by: d58e41f466b2b8ce6d51888d1648f6edb0474593

Thanks

closed 2015-10-06 by christiaanb