clash-lang/clash-compiler #84: How to enable RTS options?

wyager, 2015-10-10:
When I try to compile a program with certain RTS options, CLaSH complains about dynamic linking being required. I'm not really sure what to make of the given instructions. Any tips?

```
: clash -prof -auto-all  Main.hs
[ 1 of 16] Compiling Defs             ( Defs.hs, Defs.o )
<no location info>:
    Dynamic linking required, but this is a non-standard build (eg. prof).
    You need to build the program twice: once the normal way, and then
    in the desired way using -osuf to set the object file suffix.
```

wyager, 2015-10-10:
Also, for the record, this is so that I can figure out what's causing my program to emit a `<<loop>>` error, which corresponds to a self-referential asynchronous circuit (which is bad news in hardware).

christiaanb, 2015-10-10:
You need to build it twice, like this:

```
clash Main.hs
clash -osuf prof.o -hisuf prof.hi -prof -auto-all
```

Also, it might not be a combinational loop, see for example the end of [Troubleshooting section](http://hackage.haskell.org/package/clash-prelude/docs/CLaSH-Tutorial.html#g:18) of the tutorial. 

The problem is that Haskell's semantics are not a _perfect_ match for structural circuit design when _valid_ (buffered) feedback loops are involved. It has to do with the fact that pattern matching is strict. So what might be happening is that somewhere in your feedback loop, you made the entire output/state of your circuit dependent on the input, where only a part of the output is actually dependent on the input.

For example, you might have code like this:

```haskell
s = case x of { A -> (j,k); B -> (z,k }
```

which might be causing your loop exception, and you could make it disappear by writing it like this:

```haskell
s = (q,k)
q = case x of {A -> j; B -> z}
``` 

If needed, I can take a look at your code.

wyager, 2015-10-10:
Perfect, I will try that next time it comes up.

I managed to find the issue by replacing various circuit components with constant signals until the `<<loop>>` went away; it was, indeed, because I was reading from a mux that was switched (through a somewhat convoluted combinational path) by its own output.

However, I think I may have run into what you're describing earlier. The issue was basically

```
-- x depends on a,b,c while y only depends on b,c
foo a b c = (x,y)
    where
    y = b <> c
    x = a <> y

(x,y) = unbundle $ foo <$> a <*> b <*> c

a = f <$> y 
-- This should be fine, because y doesn't depend on a
```

But this caused a `<<loop>>` as well. I fixed it by separating `foo` into two separate functions.

closed 2015-10-10 by wyager