clash-lang/clash-compiler #85: Fail to generate VHDL with blockRamFile' in clash-ghc 0.6

masterdezign, 2015-10-12:
The problem appeared in one of the latest version of CLaSH.

When I use blockRamFile' like this:

```r = blockRamFile' myclk d96 myFile 0 raddr (signal False) 0```

CLaSH fails to generate VHDL:

``` 
...
Testbench generation took 0.003842s

<no location info>:
    CLaSH Error:
CLaSH.Netlist.BlackBox.Util(282): Unexpected file name
make: *** [build] Error 1```

christiaanb, 2015-10-12:
Could you tell me how 'myFile' is exactly defined in your code?

masterdezign, 2015-10-12:
It can be simply replaced by a string "Data/Test_96.txt" (nothing changes).

christiaanb, 2015-10-12:
Using `clash-0.6`, I am _unable_ to reproduce the bug for the following file:

```haskell
module BlockRamFileTest where

import CLaSH.Prelude
import CLaSH.Prelude.Explicit

type Clk10 = Clk "clk" 10

myclk :: SClock Clk10
myclk = sclock

topEntity :: Signal' Clk10 (Unsigned 7) -> Signal' Clk10 (BitVector 20)
topEntity raddr = blockRamFile' myclk d96 "Data/Test_96.txt" 0 raddr (signal False) 0
```

What version of `clash` are you using? and does the above file work for you?

masterdezign, 2015-10-12:
It is very strange, because the provided file generates VHDL:

```
.cabal-sandbox/bin/clash --vhdl BlockRamFileTest.hs 
[1 of 1] Compiling BlockRamFileTest ( BlockRamFileTest.hs, BlockRamFileTest.o )
Loading dependencies took 0.003709s
Applied 50 transformations
Normalisation took 0.008409s
Netlist generation took 0.002091s
Testbench generation took 0.000476s
Total compilation took 0.016326s
```

```
.cabal-sandbox/bin/clash --version
CAES Language for Synchronous Hardware, version 0.6 (using clash-lib, version: 0.6)
```

However, the project fails
```
.cabal-sandbox/bin/clash -clash-inline-limit=40 --vhdl Proj.hs
...
Loading dependencies took 1.5694s
Applied 8755 transformations
Normalisation took 22.972735s
Netlist generation took 1.008353s
Testbench generation took 0.004003s

<no location info>:
    CLaSH Error:
CLaSH.Netlist.BlackBox.Util(282): Unexpected file name
```

But it does not fail when I am using:
```clash --version
CAES Language for Synchronous Hardware, version 0.5.16 (using clash-lib, version: 0.5.14)```

clash -clash-inline-limit=40 --vhdl Proj.hs
...
[17 of 17] Compiling Proj             ( Proj.hs, Proj.o )
Loading dependencies took 1.558609s
Applied 8771 transformations
Normalisation took 23.843465s
Netlist generation took 1.220245s
Testbench generation took 0.004092s
Total compilation took 26.764155s
```

christiaanb, 2015-10-12:
Ah, I know what's going on... Starting with version 0.6, I decided that everything that needs 0-bits to represent is no longer considered synthesisable, which affects certain transformations. I forgot that I chose to represent strings by zero bits (as they are actually sort of illegal), so that's why it's now failing.

The next release of CLaSH is due friday the 16th of october 2015, which will fix this bug. Sorry for the inconvenience.

commit 8123ef80 (2015-10-12): No longer mark strings as 0-bit objects

Instead, make them rediculously large (2^32 bits). Marking them
as 0-bit, made them explicitly non-translatable, which stopped
strings from being inlined.

Fixes #85
closed 2015-10-12 by christiaanb