clash-lang/clash-compiler #87: Exception on ClaSH.Netlist.BlackBox

carldong, 2015-10-18:
I am using Stack LTS 3.8 on Ubuntu 14.04, amd64. When I try the `FIR.hs` example, I get this error:

    *** Exception: CLaSH.Netlist.BlackBox(191): No blackbox found for: CLaSH.Sized.Vector.fold

And there is no file generated, whether it is Verilog or VHDL

i ran clash using command

    stack exec -- clash --interactive FIR.hs

carldong, 2015-10-18:
OK, reading more carefully about the output, I get 

```
<interactive>:3:5:
    Couldn't match type ‘n * m’ with ‘n0 * m0’
    NB: ‘*’ is a type function, and may not be injective
    The type variables ‘n0’, ‘m0’ are ambiguous
    Expected type: [Char] -> Vec (n * m) a
      Actual type: [Char] -> Vec (n0 * m0) a
    When checking that ‘single_quote’ has the inferred type
      single_quote :: forall (n :: Nat) (m :: Nat) a.
                      [Char] -> Vec (n * m) a
    Probable cause: the inferred type is ambiguous

<interactive>:3:29:
    Couldn't match expected type ‘Vec n (Vec m a)’
                with actual type ‘[[Char]]’
    Relevant bindings include
      single_quote :: [Char] -> Vec (n * m) a
        (bound at <interactive>:3:5)
    In the first argument of ‘concat’, namely ‘["'", s, "'"]’
    In the expression: concat ["'", s, "'"]
    In an equation for ‘single_quote’:
        single_quote s = concat ["'", s, "'"]

<interactive>:9:27: Not in scope: ‘single_quote’

<interactive>:1:64: Not in scope: ‘simple_shell_escape’

<interactive>:1:71: Not in scope: ‘simple_shell_escape’
[1 of 1] Compiling FIR              ( FIR.hs, interpreted ) [GHC.TypeLits.Normalise changed]
Ok, modules loaded: FIR.
```

when I was loading `FIR.hs`

christiaanb, 2015-10-18:
Sorry for the inconvenience, the CLaSH website is set up with the expectation that you followed the "official" installation instructions and are running the latest version of the compiler. I'm quite sure that the version of the clash compiler that comes with LTS Haskell 3.8 is the 0.5 branch, where the example on the website only works with the 0.6 branch.

https://raw.githubusercontent.com/clash-lang/clash-compiler/dda6f16252674b4719a90771adca9f0591321203/examples/FIR.hs should work with the older version of clash. Also note that I have never used stack, so if it still doesn't work I won't be able to help you out until I'm back in the office on Monday and try out stack.

carldong, 2015-10-18:
The old one seems to generate files (probably)correctly, although I still get some error message:

```
<interactive>:3:5:
    Couldn't match type ‘n * m’ with ‘n0 * m0’
    NB: ‘*’ is a type function, and may not be injective
    The type variables ‘n0’, ‘m0’ are ambiguous
    Expected type: [Char] -> Vec (n * m) a
      Actual type: [Char] -> Vec (n0 * m0) a
    When checking that ‘single_quote’ has the inferred type
      single_quote :: forall (n :: Nat) (m :: Nat) a.
                      [Char] -> Vec (n * m) a
    Probable cause: the inferred type is ambiguous

<interactive>:3:29:
    Couldn't match expected type ‘Vec n (Vec m a)’
                with actual type ‘[[Char]]’
    Relevant bindings include
      single_quote :: [Char] -> Vec (n * m) a
        (bound at <interactive>:3:5)
    In the first argument of ‘concat’, namely ‘["'", s, "'"]’
    In the expression: concat ["'", s, "'"]
    In an equation for ‘single_quote’:
        single_quote s = concat ["'", s, "'"]

<interactive>:9:27: Not in scope: ‘single_quote’

<interactive>:1:64: Not in scope: ‘simple_shell_escape’

<interactive>:1:71: Not in scope: ‘simple_shell_escape’
```

christiaanb, 2015-10-19:
Ah, I get what's going on. Either you created, or `stack` created, a `.ghci` file (either in `$HOME` or in the current directory) containing at least the following:

```haskell
-- Surround a string in single quotes.
let single_quote s = concat ["'", s, "'"]
 
-- Escape a single quote in the shell. (This mess actually works.)
let escape_single_quote c = if c == '\'' then "'\"'\"'" else [c]
 
-- Simple heuristic to escape shell command arguments.
let simple_shell_escape = single_quote . (concatMap escape_single_quote)

:def hoogle \x -> return $ ":!hoogle --color " ++ (simple_shell_escape x)
:def doc \x -> return $ ":!hoogle --info --color " ++ (simple_shell_escape x)
```

What's happening, is that the `clash` binary, is actually just `ghc` with:

* Some extra commands: `:vhdl`, `:verilog`, `:systemverilog`
* Some extra modes: `--vhdl`, `--verilog`, `systemverilog`
* Some extra `LANGUAGE` extensions loaded by default.
* An alternative prelude: [CLaSH.Prelude](http://hackage.haskell.org/package/clash-prelude/docs/CLaSH-Prelude.html)

The problem in this case is that I forgot that the `ghc` binary tries to load definitions from the `.ghci` file in `--interactive` mode, and the the `CLaSH.Prelude` has definitions for `(++)` and `concat` that are different from the normal `Prelude`:

```haskell
(++) :: Vec n a -> Vec m a -> Vec (n + m) a
concat :: Vec n (Vec m a) -> Vec (n * m) a
```

So, to work-around the issue:

* If you have the `.ghci` in your `$HOME` directory, copy it to the directory from where you run the `stack exec` command.
* Update the `.ghci` file in the directory from which you run the `stack exec` command to:

  ```haskell
  import qualified Data.List as List

  -- Surround a string in single quotes.
  let single_quote s = List.concat ["'", s, "'"]

  -- Escape a single quote in the shell. (This mess actually works.)
  let escape_single_quote c = if c == '\'' then "'\"'\"'" else [c]

  -- Simple heuristic to escape shell command arguments.
  let simple_shell_escape = single_quote . (concatMap escape_single_quote)

  :def hoogle \x -> return $ ":!hoogle --color " List.++ (simple_shell_escape x)
  :def doc \x -> return $ ":!hoogle --info --color " List.++ (simple_shell_escape x)
  ```

I the next release of `clash` I will make sure it stops loading `.ghci` files, and load `.clashi` files instead.

Thanks for the bug report.

commit 896b1317 (2015-10-20): Read '.clashi' instead of '.ghci' files

Fixes #87
closed 2015-10-20 by christiaanb