clash-lang/clash-compiler #90: Phantom type causes netlist generation failure

wyager, 2015-10-23:
Got a good one for you!

So in Lambda16 I have a `Fetched` type, which is an instruction as well as some context (the PC it came from and the PC the predictor thought was coming up next). I figured that for some extra static safety, I could tag these `Fetched` values with a phantom type indicating the CPU stage they came from. E.g. an instruction + context emitted from the eXecute stage has type `Fetched X`, while one from Decode has type `Fetched D`.

The error

```
<no location info>:
    CLaSH Error:
CLaSH.Netlist(169): Not in normal form: RHS of case-projection is not a variable: case subjLet_2532294 of GHC.Types.False -> CPU.Ops.Nop
```

 started showing up in [commit 36ff605](https://github.com/wyager/Lambda16/commit/36ff60536ab3ad5ed300fe41f67966575f6584c9). As you can see, I just added the type tags and wrote a function, `pass`, which "promotes" a `Fetched` value from one stage to the next.

Note that compiling to Haskell and running it works fine; it's just netlist generation that messes up.

I suspect it has something to do with all the magic I'm using in https://github.com/wyager/Lambda16/blob/36ff60536ab3ad5ed300fe41f67966575f6584c9/CPU/Safety/Stages.hs

--Will

closed 2015-10-23 by christiaanb

christiaanb, 2015-10-23:
Thanks for the bug report. Of course it turned out to be yet another bug in the recently introduced DEC transformation.

wyager, 2015-10-23:
Great, thanks. This is why I love open source; the bugfix time on some projects is measured in hours!

christiaanb, 2015-10-24:
The fix is included as part of (the just released) version 0.6.3 of the compiler. I really really hope this was the last bug in the DEC transformation.

wyager, 2015-10-24:
Well I'll keep trying to break it :)

Cheers!

commit 9db4e8ff (2015-10-23): DEC: don't generate single-branch case-expressions

Fixes #90