clash-lang/clash-compiler #91: The order of "case" in verilog

Alaya-in-Matrix, 2015-10-30:
I am trying to display numbers on segment LED, so I write such a encoder:

```haskell
segEncoder :: (Integral a) => a -> SegDisp
segEncoder n = case n of
                 0 -> 0b00111111
                 1 -> 0b00000110
                 2 -> 0b01011011
                 3 -> 0b01001111
                 4 -> 0b01100110
                 5 -> 0b01101101
                 6 -> 0b01111101
                 7 -> 0b00000111
                 8 -> 0b01111111
                 9 -> 0b01101111
                 _ -> 0b00111111

segV = map segEncoder
```

but when I tried to encode `Unsigned 3` numbers, for number 0, the displayed number is 8. so I looked at the compiled verilog code:

```verilog
// Automatically generated Verilog-2001
module SegLED_segEncoder_1(ds_i1
                          ,topLet_o);
  input [3:0] ds_i1;
  output [7:0] topLet_o;
  reg [7:0] topLet_o_reg;
  always @(*) begin
    case(ds_i1)
      4'd9 : topLet_o_reg = 8'd111;
      4'd8 : topLet_o_reg = 8'd127;
      4'd7 : topLet_o_reg = 8'd7;
      4'd6 : topLet_o_reg = 8'd125;
      4'd5 : topLet_o_reg = 8'd109;
      4'd4 : topLet_o_reg = 8'd102;
      4'd3 : topLet_o_reg = 8'd79;
      4'd2 : topLet_o_reg = 8'd91;
      4'd1 : topLet_o_reg = 8'd6;
      4'd0 : topLet_o_reg = 8'd63;
      default : topLet_o_reg = 8'd63;
    endcase
  end
  assign topLet_o = topLet_o_reg;
endmodule
```

Why is the order in "case" from 9 to 0, instead of from 0 to 9, which matches the semantic of haskell more?

christiaanb, 2015-10-30:
When GHC goes from Haskell to its internal Core language, it puts the default pattern, `_` at the top (I cannot remember right now why it does that). I subsequently translate `_` in Core, to `default` in Verilog. However, if I would keep the order the same as it was given to me by GHC, the Verilog would always match against the `default` pattern, as it would be on top. That would of course be wrong. So what I do is simply reverse the alternatives that GHC gives me, hence resulting in correct Verilog, with `default` at the bottom.

Also note that, in Haskell, all patterns are unique, so the order in which they are put (with the exception of the default pattern) doesn't matter.

christiaanb, 2015-10-31:
Ah, I didn't realize the problem yesterday as I had a long day of travel behind me. I'm on holiday right now, and will be back on the 11th of November. I won't be able to fix it until then.

Perhaps you want to try and supply a patch? Here are some instructions on how to: https://github.com/clash-lang/clash-compiler/wiki/Getting-started

The code that needs patching is probably:  https://github.com/clash-lang/clash-compiler/blob/master/clash-lib/src/CLaSH/Netlist.hs#L216
Remember though about what I said that GHC puts the default pattern at the front, and it must be put at the end.

commit 02163a17 (2015-11-11): Stop reversing alts list in netlist generation

We used to do this because GHC puts the default alternative in
the first position. Simply reversing the list of alternatives
is would make VHDL/Verilog be the same as the original Haskell,
under the assumption that all patterns are unique. This however
isn't true for literal patterns, where the scrutinee might not
have enough bits to represent all te alternatives. So what
we do now is just put the Default pattern at the end.

Fixes #91
commit fb121db0 (2015-11-17): foldr label must be unique

Fixes #91
closed 2015-11-11 by christiaanb