clash-lang/clash-compiler #92: Compilation takes a long time/forever with more specialized type

wyager, 2015-11-02:
I was investigating some approaches for improving the simplicity of the generated verilog, and I was curious if further type specialization would result in the generation of fewer modules. See https://github.com/wyager/Lambda16/commit/080ec13eba2a312296088b7e3df45813a3558114

For whatever reason, this change causes Cλash to have a hard time. It's either taking a very long time or not terminating at all.

I had similar problems when I removed the WriteCache arguments altogether and re-wrote the function in terms of signals, such that the WriteCaches were generated within the function instead of passed as an argument.

It seems like this only happens when I specialize the size of the caches within the same module as the cache-size-parametric functions.

christiaanb, 2015-11-02:
Thanks for the report. I'm on vacation right now, and won't be able to work on a fix until November 11.

wyager, 2015-11-02:
No problem. Should be able to work around it until then. Have fun.

closed 2015-11-11 by christiaanb

christiaanb, 2015-11-11:
The current fix really feels like it is a red hearing, because I cannot really put my finger on to why the old code was causing problems. I will investigate further tomorrow.

Regardless, your code does compile now.

commit 1a0bade5 (2015-11-11): DEC: root of the case-tree must contain at least 2 alternatives

Before, a case-tree was disjoint when it has more that 2 branches,
or, if it has a single branch which is itself disjoint. We do
this so we can look through pattern matches on product types to
find disjoin applications. However, we must _not_ do this for the
root of the case-tree.

Fixes #92