clash-lang/clash-compiler #95: FIR example fails to generate VHDL / Verilog files.

VitorCBSB, 2015-11-22:
Hey there, I was interested in using CLaSH so I followed the verification part up to step 3, namely `Execute, in the interpreter, the :vhdl command`, where it then fails.

I installed CLaSH by cloning the repo then `stack build` ing it, so I might have missed some crucial step. I'm running Windows 10, if that helps anything.

If it also helps, other examples, such as MAC.hs and Calculator.hs do generate the HDL files.

```
C:\Users\Vitor Coimbra\Test\src>stack exec -- clash --interactive Main.hs
CLaSHi, version 0.5.15 (using clash-lib, version 0.5.13):
http://www.clash-lang.org/  :? for help
clash.exe: warning: _tzset from msvcrt is linked instead of __imp__tzset
[1 of 1] Compiling FIR              ( Main.hs, interpreted )
Ok, modules loaded: FIR.
*FIR> :verilog
[1 of 1] Compiling FIR              ( Main.hs, Main.o )
Loading dependencies took 0.1879401s
Applied 211 transformations
Normalisation took 0.2263861s
*** Exception: CLaSH.Netlist.BlackBox(191): No blackbox found for: CLaSH.Sized.Vector.fold
*FIR>
```

christiaanb, 2015-11-22:
I don't use the stack tool myself, but I guess I really need to check it out. I'll do so first thing when I get in the office tomorrow.

christiaanb, 2015-11-22:
Ah, I see what's going on. Stack installed the 0.5 branch of clash (0.5 branch is the branch on LTS 3), however, the FIR example can only be used with the latest 0.6 branch. Now, you either set the stack resolver to stackage nightly, and install the 0.6 branch, or use this older version of the FIR example: https://github.com/clash-lang/clash-compiler/blob/dda6f16252674b4719a90771adca9f0591321203/examples/FIR.hs

Ericson2314, 2015-11-22:
Hmm, according to https://github.com/commercialhaskell/stack/blob/master/doc/yaml_configuration.md and our `stack.yaml`, it ought to be built the checked out version, not the LTS version--i.e. this stack file should be appropriate for developing CLaSH itself.

Arguably users of CLaSH should write their own stack/cabal files using the hackage releases / LTS for their projects, but regardless seems weird that this arose in the first place.

VitorCBSB, 2015-11-22:
Ah, I see now. Yes, that works. The main problem is my ignorance with the stack tool.

I first tried to install CLaSH with a `stack install clash-ghc`, which installs it "globally"-ish. Facing this error, I cloned the repo and built it with `stack build`, creating the clash binary in the local project only. I completely forgot about that detail and tried to use it in another project and the result is that it was still using the first one I installed.

Thanks for your help.

closed 2015-11-22 by VitorCBSB

christiaanb, 2015-11-23:
Glad the hear that everything was resolved. I will still investigate the `stack` tool in more detail so I can be of more help in the future, and also update existing documentation.

christiaanb, 2015-11-24:
I've updated the FAQ to include some information on how to install and use `clash` using `stack`: https://github.com/clash-lang/clash-compiler/wiki/FAQ#how-do-i-run-and-install-clash-using-stack