clash-lang/clash-compiler #96: Stale files in vhdl folder

ggreif, 2015-11-25:
When iterating a design sometimes stale files remain in `vhdl/*` (and other) folders. It would be nice to have an extension to the synthesis commands `:vhdl` that clean out stale files.
Would `:vhdl-clean` or `:vhdl -clean` work?

christiaanb, 2015-11-25:
We used to do this by default, but somehow this feature got dropped when we switched to supporting multiple backends. I prefer that cleaning up would be the default again.

So in interactive mode we would have: `:vhdl [noclean]`, where the square brackets indicate that it is an optional argument.

In batch mode we would have `-clash-noclean` flag. So you'd say: `clash --vhdl -clash-noclean <MODULE.hs>`.

I'll implement this soon, probably by tomorrow already.

ggreif, 2015-11-25:
woot!

christiaanb, 2015-11-26:
OK, slight change of what I said earlier. There's one static flag `-clash-noclean`, which tells CLaSH _NOT_ to clean up existing HDL files, and it applies to both the interactive as batch modes. That is, there is no additional option for the interactive `:vhdl` command.

Being a static flag, it can only be supplied when you start the `clash` compiler. So you _cannot_ do `:set -fclash-noclean` in interactive mode.

commit 629ac27b (2015-11-26): Remove existing HDL files before generating new ones

Fixes #96
closed 2015-11-26 by christiaanb