clash-lang/clash-compiler #97: Normalizer chokes on expectedOutput

ggreif, 2015-11-27:
``` haskell
{-# LANGUAGE MultiParamTypeClasses, FlexibleInstances, ScopedTypeVariables, LambdaCase, DefaultSignatures #-}

module CAM where

import CLaSH.Prelude

import Data.Monoid
import Unsafe.Coerce (unsafeCoerce)

class BlockRamAssoc k v where
  type Bucket k v
  type Bucket k v = Vec 2 (Maybe (k, v))
  initBuckets :: KnownNat (2 ^ n) => k -> v -> Vec (2 ^ n) (Bucket k v)
  findColumn :: k -> (Bucket k v) -> Maybe v
  default findColumn :: Eq k => k -> Vec (n + 1) (Maybe (k, v)) -> Maybe v
  findColumn k = fold pick . map (\case Just (k', v) | k == k' -> Just v; _ -> Nothing)
    where pick cool@(Just _) _ = cool
          pick _ other = other

cam :: forall k v n . (BlockRamAssoc k v, KnownNat n, KnownNat (2 ^ n)) => (k -> Unsigned n) -> Signal k -> Signal (Maybe v)
cam f keys = findColumn <$> keys <*> ram
  where hashes = f <$> keys
        ram = blockRamPow2 (initBuckets (undefined::k) (undefined::v)) hashes hashes (pure False) undefined

instance BlockRamAssoc Bool Bool where
  initBuckets _ _ = unsafeCoerce fakeBuckets

fakeBuckets :: Vec (2 ^ 6) (Bucket Bool Bool)
fakeBuckets = empty :> empty :> (Nothing :> Just (True, False) :> Nil) :> repeat empty
    where empty = repeat Nothing

topEntity :: Signal Bool -> Signal (Maybe Bool)
topEntity = cam hashfunc
  where hashfunc True = 2 :: Unsigned 6
        hashfunc _ = 0

testInput :: Signal Bool
testInput = stimuliGenerator (False :> True :> True :> False :> False :> False :> True :> True :> False :> Nil)

expectedOutput :: Signal (Maybe Bool) -> Signal Bool
expectedOutput = outputVerifier $ Nothing :> Nothing :> Just False :> Nothing :> Nothing :> Nothing :> Nothing :> Just False :> Nothing :> Nothing :> Nil

{-
>>> Prelude.drop 1 (sampleN 16 (expectedOutput $ topEntity testInput))
-}
```

I get this error from the normalizer:

```
...
Applied 51 transformations
CLaSH.Normalize(118): Expr belonging to bndr: GHC.Show.showList___showl27224 (:: GHC.Types.[] GHC.Types.Char
-> GHC.Types.[] (GHC.Base.Maybe GHC.Types.Bool)
-> GHC.Types.[] GHC.Types.Char) remains recursive after normalization:
 (lvl18 :: GHC.Types.[] GHC.Types.Char) ->
 (ds2 :: GHC.Types.[] (GHC.Base.Maybe GHC.Types.Bool)) ->
letrec
  repANF1 :: GHC.Types.[] GHC.Types.Char
  = GHC.Show.showList___showl27224
      lvl18
      ys
  repANF :: GHC.Types.[] GHC.Types.Char
  = GHC.Show.$fShowMaybe_$cshowsPrec
    @GHC.Types.Bool
      GHC.Show.$fShowBool1914257750
      (GHC.Types.I#
         0)
      y
      repANF1
  altLet :: GHC.Types.[] GHC.Types.Char
  = GHC.Types.:
    @GHC.Types.Char
      GHC.Show.showList__11914257943
      repANF
  y :: GHC.Base.Maybe GHC.Types.Bool
  = case ds2 of
      GHC.Types.:
        (sel :: GHC.Base.Maybe GHC.Types.Bool)
        (wild :: GHC.Types.[] (GHC.Base.Maybe GHC.Types.Bool)) ->
        sel
  ys :: GHC.Types.[] (GHC.Base.Maybe GHC.Types.Bool)
  = case ds2 of
      GHC.Types.:
        (wild :: GHC.Base.Maybe GHC.Types.Bool)
        (sel :: GHC.Types.[] (GHC.Base.Maybe GHC.Types.Bool)) ->
        sel
  topLet :: GHC.Types.[] GHC.Types.Char
  = case ds2 of
      GHC.Types.[] -> lvl18
      GHC.Types.:
        (y1 :: GHC.Base.Maybe GHC.Types.Bool)
        (ys1 :: GHC.Types.[] (GHC.Base.Maybe GHC.Types.Bool)) ->
        altLet
in topLet
Applied 137 transformations
Testbench generation took 0.321429s
Total compilation took 0.853051s
```

christiaanb, 2015-11-27:
That message is actually sort of a warning, though I guess it is not really clear. The problem is that `outputVerifier` uses the `assert` function, and the `assert` function uses `show`. The `show` member of the `Show` instance for lists is recursive, and that's what the compiler is complaining about.

Normally, the above _warning_ would lead to an error saying: `Callgraph after normalisation contains following recursive cycles: ...`, however, because `assert` is a primitive, the `show` member in the `Show` dictionary passed to `assert` is never used. Consequently, its "OK" that `show` is remains recursive after normalisation.

So yeah, sometimes when you see that message, it will result in the compiler in failing to translate your Haskell to VHDL/Verilog. However, as you've probably noticed in your case, the compiler was still able to translate yoru Haskell to VHDL/Verilog as the recursive function ended up never being used.

ggreif, 2015-11-27:
@christiaanb wow, I did not even dare to look into the VHDL output after that scary message. And yes, I loaded it into Vivado, and it was happy with it in both simulation and synthesis mode. I am closing this, but maybe something should be done with the scary message.

closed 2015-11-27 by ggreif

christiaanb, 2015-11-27:
I could suppress the message, and make it so that it is only reported when debug output is turned on? Or should I always leave it on, but change the wording a little bit (and perhaps only show the complete expression when debug output is turned on)?

ggreif, 2015-11-27:
Is it possible to detect whether the recursive function *actually*
ends up in the RTL? Is the outputter some write-monad-like
construction?

christiaanb, 2015-11-27:
It is possible to detect whether recursive functions would end up in RTL, or not. Again, primitives pose a bit of a problem, as some primitives, like the one for `map`, actually use their function argument. Sadly, the warning is implemented as a simple `Debug.Trace.trace`, and hence the warning is printed immediately as it is found.