clash-lang/clash-compiler #98: Testbench suppresses first cycle

ggreif, 2015-11-27:
Maybe this happens only when `BlockRam` is involved, but I miss the first cycle from the simulation for this test input:
```
{-# LANGUAGE MultiParamTypeClasses, FlexibleInstances, ScopedTypeVariables, LambdaCase, DefaultSignatures #-}

module CAM where

import CLaSH.Prelude

class BlockRamAssoc k v where
  type Bucket k v
  type Bucket k v = Vec 2 (Maybe (k, v))

  initBuckets :: KnownNat (2 ^ n) => k -> v -> Vec (2 ^ n) (Bucket k v)
  default initBuckets :: KnownNat (2 ^ n) => k -> v -> Vec (2 ^ n) (Vec 2 (Maybe (k, v)))
  initBuckets _ _ = repeat (repeat Nothing)

  findColumn :: k -> (Bucket k v) -> Maybe v
  default findColumn :: Eq k => k -> Vec (n + 1) (Maybe (k, v)) -> Maybe v
  findColumn k = fold pick . map (\case Just (k', v) | k == k' -> Just v; _ -> Nothing)
    where pick cool@(Just _) _ = cool
          pick _ other = other

  buildBucket :: (k, v) -> Bucket k v

lcam :: forall k v n . (BlockRamAssoc k v, KnownNat n, KnownNat (2 ^ n)) => (k -> Unsigned n) -> Signal (Either k (k, v)) -> Signal (Maybe v)
lcam f moves = findColumn <$> keys <*> ram
  where hashes = f <$> keys
        keys = (\case Left k -> k; Right (k, _) -> k) <$> moves
        writes = (\case Left _ -> False; Right _ -> True) <$> moves
        values = (\case Left k -> undefined; Right kv -> buildBucket kv) <$> moves
        ram = blockRamPow2 (initBuckets (undefined::k) (undefined::v)) hashes hashes writes values

instance BlockRamAssoc Bool Bool where
  buildBucket = repeat . Just -- FOR NOW!

topEntity :: Signal (Either Bool (Bool, Bool)) -> Signal (Maybe Bool)
topEntity = lcam hashfunc
  where hashfunc True = 2 :: Unsigned 6
        hashfunc _ = 0

testInput :: Signal (Either Bool (Bool, Bool))
testInput = stimuliGenerator (l' :> l' :> r False :> r True :> r True :> r False :> l :> r False :> r False :> r True :> r True :> r False :> Nil)
  where r = Left -- retrieve
        l = Right (False, True) -- learn False
        l' = Right (True, False) -- learn True

expectedOutput :: Signal (Maybe Bool) -> Signal Bool
expectedOutput = outputVerifier $ Nothing :> Nothing :> Nothing :> Nothing :> Just False :> Nothing :> Nothing :> Nothing :> Just True :> Nothing :> Just False :> Nothing :> Just True :> Nil

{-
>>> Prelude.drop 1 (sampleN 16 (expectedOutput $ topEntity testInput))
-}
```

christiaanb, 2015-11-27:
Can you expand on what you mean by "miss the first cycle"? Are all expected outputs shifted 1 cycle back in time?

ggreif, 2015-11-27:
Right, the first cycle does not appear in the waveform for the
simulation. I had to prepend a cycle to both `testInput` and
`expectedOutput` so that I could finally see an initial learning
cycle.

christiaanb, 2015-11-30:
Well... you have to remember that the first output, is the output defined by the reset... which the blockram does not get. Take a look at: https://github.com/clash-lang/clash-compiler/blob/master/tests/shouldwork/Signal/BlockRamFile.hs

Specifically as to how we handle the undefined value:

```haskell
zeroAt0 :: Signal (Unsigned 8) -> Signal (Unsigned 8)
zeroAt0 a = mux en a 0
  where
    en = register False (signal True)

topEntity :: Signal (Unsigned 8) -> Signal (Unsigned 8)
topEntity rd = zeroAt0 (unpack <$> dout)
  where
    dout = blockRamFilePow2 "memory.list" 0 rd (signal False) 0
```

Does applying a similar technique to your specification offer expected VHDL simulation results?

ggreif, 2015-12-01:
@christiaanb I didn't express myself precisely enough. The Haskell simulation properly shows the first cycle. What is (was) missing a cycle is the Vivado waveform display for "Behavioral Simulation". After prepending an extra (redundant) cycle to the Haskell simulation the corresponding waveform appeared, but again with a width of only one period instead of two. I guess you have no access to the Xilinx tools, so maybe you can try it with Altera?

christiaanb, 2015-12-02:
Ah, I see where the misunderstanding comes from. You have 13 testinputs and 13 expected outputs, but you are only seeing 12 cycles. This is correct. The first testinput and expected output are applied when the design comes out of reset, but before the first rising edge of the clock. This happens between 2ns and 3ns. The remaining 12 are then applied on the 12 clock cycles.

christiaanb, 2015-12-30:
@ggreif can this issue be closed?

ggreif, 2015-12-30:
Yes. It is understood, and I can see the wiggles around 2-3 ns in the
waveform now. Shouldn't always have my bird-view glasses on :-)

Thanks!

closed 2015-12-30 by ggreif