clash-lang/clash-compiler #105: How to make Cλash compilation verbose?

wyager, 2015-12-23:
`clash --verilog` is failing to terminate on `Hardware.hs` [here](https://github.com/wyager/HaSKI/tree/61667bfd438ffed64aac223b0d2b2eb1047797fb).

I tried `clash --verilog -v`, but this just makes the GHC steps verbose. How do I make the hardware steps verbose?

christiaanb, 2015-12-23:
"-clash-debug DebugName" will give you the information as to whether the compiler is stuck on a single transformation, or looping between a series of transformations.

wyager, 2015-12-23:
Got it.

I'll open a new issue with the bug, but there was also some other weirdness over the course of debugging it...

For example, I used a Word to count the number of things in a stack, and for whatever reason, GHC.Prim.Word confused Clash. Clash thought something was recursive even though it was a perfectly synthesizable knot-tied recursion. I had to break the knot-tying to get the actual error, which was something along the lines of "GHC.Prim.W# is empty" or something.

closed 2015-12-23 by wyager