clash-lang/clash-compiler #106: Exponent literals cause Clash not to terminate

wyager, 2015-12-23:
Take a look at https://github.com/wyager/HaSKI/commit/a7bfee53128e55660a6a27f0b230bbe5c55bfbd0

Those `0x...` literals used to be `2^20` and `2^21`. Looks like clash got stuck in a loop of trying to inline `step2` and `default_...` until I changed them to hex literals.

comonoidial, 2015-12-23:
I guess the problem here is that the unspecialized version of (^) is used, as defined by ghc:
```haskell
(^) :: (Num a, Integral b) => a -> b -> a
x0 ^ y0 | y0 < 0    = error "Negative exponent"
        | y0 == 0   = 1
        | otherwise = f x0 y0
    where -- f : x0 ^ y0 = x ^ y
          f x y | even y    = f (x * x) (y `quot` 2)
                | y == 1    = x
                | otherwise = g (x * x) ((y - 1) `quot` 2) x
          -- g : x0 ^ y0 = (x ^ y) * z
          g x y z | even y = g (x * x) (y `quot` 2) z
                  | y == 1 = x * z
                  | otherwise = g (x * x) ((y - 1) `quot` 2) (x * z)
````
This seems quite a nasty recursive structure to eliminate by inlining.

wyager, 2015-12-25:
The interesting thing is that Clash seemed to keep switching between optimizing the `Stack` and `Heap` constructors and optimizing the function that made use of them... I would figure if it got stuck in an optimizing loop it would be local to the site of `(^)`.

christiaanb, 2015-12-30:
Does compilation terminate if you make the power smaller? e.g. `2^4`?

wyager, 2016-01-04:
Weird. Looks like up to `2^8` works fine. `2^9` or higher breaks.

martijnbastiaan, 2019-07-31:
Fixed after merging https://github.com/clash-lang/clash-compiler/pull/592.

closed 2019-07-31 by martijnbastiaan