clash-lang/clash-compiler #107: Stack

flip111, 2016-01-06:
I was using haskell platform with cabal before, now i removed all of this and start over with stack. How can i use this library with stack?

christiaanb, 2016-01-06:
https://github.com/clash-lang/clash-compiler/wiki/FAQ#how-do-i-run-and-install-clash-using-stack

flip111, 2016-01-06:
Ups sorry didn't notice there was a Wiki attached ! Thank you very much

flip111, 2016-01-15:
stack seems to have some issues with the requested version of the compiler

```
» stack install --resolver=nightly clash-ghc
Run from outside a project, using implicit global project config
Using resolver: nightly specified on command line
Selected resolver: nightly-2016-01-14
Downloaded nightly-2016-01-14 build plan.    
Caching build plan
Updating package index Hackage (mirrored at https://github.com/commercialhaskell/all-cabal-hashes.g                                                                                                   Fetched package index.    
Populated index cache.    
No compiler found, expected minor version match with ghc-7.10.3 (x86_64) (based on resolver setting in /home/ph88/.stack/global-project/stack.yaml).
Try running "stack setup" to install the correct GHC into /home/ph88/.stack/programs/x86_64-linux/

» stack setup
Run from outside a project, using implicit global project config
Using resolver: lts-3.19 from implicit global project's config file: /home/ph88/.stack/global-project/stack.yaml
stack will use a locally installed GHC
For more information on paths, see 'stack path' and 'stack exec env'
To use this GHC and packages outside of a project, consider using:
stack ghc, stack ghci, stack runghc, or stack exec
```

christiaanb, 2016-01-15:
So what happens if you run:

```
stack setup --resolver=nightly
stack install --resolver=nightly clash-ghc
```

christiaanb, 2016-01-15:
Ok, I just tried this on my windows machine, which doesn't have any development tools installed:

* download stack windows installer
* install stack
* open the command line
* `stack setup --resolver=nightly`
* `stack install --resolver=nightly clash-ghc`
* `stack exec --resolver=nightly -- clash --interactive`

And the CLaSH REPL started. What part in the above steps fails on your computer?

flip111, 2016-01-15:
Its working already better when i do the setup step. But my atom cpu is slow so i can respond fast :P

Right now it's missing some headers i will try to to find them. (ncursus)

By the way i'm on ubuntu 15.10.

flip111, 2016-01-15:
clash interactive started successfully. Per add the info about stack setup and libcursus to FAQ?

closed 2016-01-15 by flip111

Ericson2314, 2016-01-18:
Maybe `stack.yaml` should be changed to use the nightly resolver by default?

flip111, 2016-04-28:
I noticed that in my global config file i had lts-3.19 set. Perhaps with a newer lts version (such as `5.14`) the nightly resolver doesn't have to be used anymore.

flip111, 2016-04-28:
Hmm no, that last comment doesn't seem to be true

```
» stack install clash-ghc
Run from outside a project, using implicit global project config
Using resolver: lts-5.14 from implicit global project's config file: /home/flip111/.stack/global-project/stack.yaml
concurrent-supply-0.1.8: configure
concurrent-supply-0.1.8: build
base-orphans-0.5.4: configure
base-orphans-0.5.4: build
StateVar-1.1.0.4: configure
StateVar-1.1.0.4: build
data-default-class-0.0.1: configure
data-default-class-0.0.1: build
data-default-class-0.0.1: copy/register
base-orphans-0.5.4: copy/register
Progress: 4/53
--  While building package StateVar-1.1.0.4 using:
      /home/flip111/.stack/setup-exe-cache/x86_64-linux/setup-Simple-Cabal-1.22.5.0-ghc-7.10.3 --builddir=.stack-work/dist/x86_64-linux/Cabal-1.22.5.0 build --ghc-options " -ddump-hi -ddump-to-file"
    Process exited with code: ExitFailure 1
    Logs have been written to: /home/flip111/.stack/global-project/.stack-work/logs/StateVar-1.1.0.4.log

    Configuring StateVar-1.1.0.4...
    Building StateVar-1.1.0.4...
    Preprocessing library StateVar-1.1.0.4...
    
    /tmp/stack17658/StateVar-1.1.0.4/src/Data/StateVar.hs:80:8:
        Could not find module ‘Control.Concurrent.STM’
        There are files missing in the ‘stm-2.4.4.1@stm_EyxFXFfnfY05IrjGaqCxeD’ package,
        try running 'ghc-pkg check'.
        Use -v to see a list of the files searched for.

--  While building package concurrent-supply-0.1.8 using:
      /home/flip111/.stack/setup-exe-cache/x86_64-linux/setup-Simple-Cabal-1.22.5.0-ghc-7.10.3 --builddir=.stack-work/dist/x86_64-linux/Cabal-1.22.5.0 build --ghc-options " -ddump-hi -ddump-to-file"
    Process exited with code: ExitFailure 1
    Logs have been written to: /home/flip111/.stack/global-project/.stack-work/logs/concurrent-supply-0.1.8.log

    Configuring concurrent-supply-0.1.8...
    Building concurrent-supply-0.1.8...
    Preprocessing library concurrent-supply-0.1.8...
    
    /tmp/stack17658/concurrent-supply-0.1.8/src/Control/Concurrent/Supply.hs:42:8:
        Could not find module ‘Data.Hashable’
        There are files missing in the ‘hashable-1.2.4.0@hasha_EalUWgp8NarBYWPbvYAD8I’ package,
        try running 'ghc-pkg check'.
        Use -v to see a list of the files searched for.
```

flip111, 2016-07-07:
To let you know ...
```
sudo apt-get install libncurses5-dev
stack install clash-ghc
```
is enough now on ubuntu 16.04
it will use lts-6.3, don't need to use nightly anymore

christiaanb, 2016-07-07:
Thanks for sharing.