clash-lang/clash-compiler #109: Exception: CLaSH.Netlist(285): Unknown function: a3_227

ggreif, 2016-01-07:
While compiling my CAM design I found two CLaSH bugs. This is number two. I get
```
Loading dependencies took 0.449427s
Applied 268 transformations
Normalisation took 0.819746s
*** Exception: CLaSH.Netlist(285): Unknown function: a3_227
```
when compiling the attached file.

Writing the local function `both` with a catch-all pattern removes the problem.
[CAM.txt](https://github.com/clash-lang/clash-compiler/files/81564/CAM.txt)

christiaanb, 2016-01-07:
I'm surprised your code works at all: you're storing functions in registers, right? I don't think the CLaSH compiler can handle that...

ggreif, 2016-01-07:
Funny thing, it works! Just uncomment that line and it even generates VHDL. The power of defunctionalisation?

christiaanb, 2016-01-07:
I wonder if it is meaning-preserving.... does the VHDL even work?

ggreif, 2016-01-08:
I'll test it tomorrow.

ggreif, 2016-01-08:
I get may errors from Vivado like this:
```
[Synth 8-2139] illegal identifier : __VOID__ ["vhdl/CAM/CAM_cam_zdscam_1.vhdl":123]
```
when synthesizing `:vhdl` from this:
``` haskell
cam :: forall k v n . (BlockRamAssoc k v, KnownNat n, KnownNat (2 ^ n)) => (k -> Unsigned n) -> Signal (Either k (k, v)) -> Signal (Maybe v)
cam f move = condUpdater ram out wr hash
  where ram = condSigWrite $ maybeWrite $ {- BUG #2 FIXME: readNew -}id (blockRamPow2 (initBuckets (undefined::k) (undefined::v)))
        out = const Nothing `register` (findColumn <$> key)
        wr :: Signal (Maybe v -> Maybe (Bucket k v -> Bucket k v))
        wr = liftA both (Nothing `register` pair)
        both (Just kv) Nothing = Just (augmentBucket kv)
        both _ _ = Nothing -- WORKS!
        --both _ Just{} = Nothing
        --both Nothing _ = Nothing -- BUG #1 : *** Exception: CLaSH.Netlist(285): Unknown function: a3_227
        hash :: Signal (Unsigned n)
        hash = f <$> key
        key = (\case Left k -> k; Right (k, _) -> k) <$> move
        pair = (\case Left _ -> Nothing; Right p -> Just p) <$> move
```

ggreif, 2016-01-08:
Looks like the __VOID__ identifier comes from https://github.com/clash-lang/clash-compiler/blob/master/clash-lib/src/CLaSH/Netlist/BlackBox.hs#L100 and/or https://github.com/clash-lang/clash-compiler/blob/master/clash-lib/src/CLaSH/Netlist/BlackBox.hs#L119

christiaanb, 2016-01-08:
Yeah, that basically means that the compiler wasn't able to translate to VHDL. The compiler should error out instead of generating bogus VHDL.

Also, the compiler doesn't actually have a defunctionalisation pass. It has a higher-order specialization pass to get rid of the higher-order values. I guess I should also add a defunctionalisation pass to get rid of the remaining higher-order values.

ggreif, 2016-01-08:
Good to hear that this is not a fundamental problem. This is a great
paper w.r.t. defunctionalisation:
http://www.diku.dk/~paba/pubs/entries/hutton16wf.html

I would love to be able to have signals of function type around. That
is where the power of FP comes from, after all!

I have a free weekend ahead, I can sink my teeth in it, too.

ggreif, 2016-01-08:
But the original bug is not exactly the `__VOID__` issue. There is something subtle going on with the three-clause pattern match. Both may be related, though.

christiaanb, 2016-01-08:
They are related, but I don't have the time to explain as I'm attending the Dutch functional programming day. I'll elaborate tomorrow.

christiaanb, 2016-01-10:
OK, so the haskell: 

```haskell
both (Just kv) Nothing = Just (augmentBucket kv)
both _ _ = Nothing
```

would be desugared to the following Core (by GHC):

```
both_a4XO
  :: forall t_a6bI k_a6bJ v_a6bK.
     BlockRamAssoc k_a6bJ v_a6bK =>
     Maybe (k_a6bJ, v_a6bK)
     -> Maybe t_a6bI
     -> Maybe (Bucket k_a6bJ v_a6bK -> Bucket k_a6bJ v_a6bK)
[LclId, Str=DmdType]
both_a4XO =
  \ (@ t_a6bN)
    (@ k_a6bO)
    (@ v_a6bP)
    ($dBlockRamAssoc_a6bQ :: BlockRamAssoc k_a6bO v_a6bP)
    (ds_dafp :: Maybe (k_a6bO, v_a6bP))
    (ds_dafq :: Maybe t_a6bN) ->
    let {
      fail_dafr
        :: GHC.Prim.Void#
           -> Maybe (Bucket k_a6bO v_a6bP -> Bucket k_a6bO v_a6bP)
      [LclId, Str=DmdType]
      fail_dafr =
        \ _ [Occ=Dead, OS=OneShot] ->
          GHC.Base.Nothing
            @ (Bucket k_a6bO v_a6bP -> Bucket k_a6bO v_a6bP) } in
    case ds_dafp of _ [Occ=Dead] {
      __DEFAULT -> fail_dafr GHC.Prim.void#;
      Just kv_a4YA ->
        case ds_dafq of _ [Occ=Dead] {
          __DEFAULT -> fail_dafr GHC.Prim.void#;
          Nothing ->
            GHC.Base.Just
              @ (Bucket k_a6bO v_a6bP -> Bucket k_a6bO v_a6bP)
              (augmentBucket @ k_a6bO @ v_a6bP $dBlockRamAssoc_a6bQ kv_a4YA)
```

Whilst the following Haskell:

```haskell
 both (Just kv) Nothing = Just (augmentBucket kv)
 both _ Just{} = Nothing
 both Nothing _ = Nothing
```

Seems to be desugared to:

```
both_a4WP =
  \ (@ t_a6bb)
    (@ k_a6bc)
    (@ v_a6bd)
    ($dBlockRamAssoc_a6be :: BlockRamAssoc k_a6bc v_a6bd)
    (ds_daeN :: Maybe (k_a6bc, v_a6bd))
    (ds_daeO :: Maybe t_a6bb) ->
    let {
      fail_daeU
        :: GHC.Prim.Void#
           -> Maybe (Bucket k_a6bc v_a6bd -> Bucket k_a6bc v_a6bd)
      [LclId, Str=DmdType]
      fail_daeU =
        \ _ [Occ=Dead, OS=OneShot] ->
          case ds_daeO of _ [Occ=Dead] {
            __DEFAULT ->
              (\ _ [Occ=Dead, OS=OneShot] ->
                 case ds_daeN of _ [Occ=Dead] {
                   __DEFAULT ->
                     (\ _ [Occ=Dead, OS=OneShot] ->
                        Control.Exception.Base.patError
                          @ (Maybe (Bucket k_a6bc v_a6bd -> Bucket k_a6bc v_a6bd))
                          "examples/CAM.hs:(69,9)-(72,32)|function both"#)
                       GHC.Prim.void#;
                   Nothing ->
                     GHC.Base.Nothing @ (Bucket k_a6bc v_a6bd -> Bucket k_a6bc v_a6bd)
                 })
                GHC.Prim.void#;
            Just _ [Occ=Dead] ->
              GHC.Base.Nothing @ (Bucket k_a6bc v_a6bd -> Bucket k_a6bc v_a6bd)
          } } in
    case ds_daeN of _ [Occ=Dead] {
      __DEFAULT -> fail_daeU GHC.Prim.void#;
      Just kv_a4XG ->
        case ds_daeO of _ [Occ=Dead] {
          __DEFAULT -> fail_daeU GHC.Prim.void#;
          Nothing ->
            GHC.Base.Just
              @ (Bucket k_a6bc v_a6bd -> Bucket k_a6bc v_a6bd)
              (augmentBucket @ k_a6bc @ v_a6bd $dBlockRamAssoc_a6be kv_a4XG)
```

So the problem with the second version is the `patError`, through a series of transformation, CLaSH ends up with the following case-statement which it can (currently) not reduce any further:

```haskell
case Control.Exception.Base.patError @ (Maybe (Bucket k_a6bc v_a6bd -> Bucket k_a6bc v_a6bd)) of
  ....
```

As a result, we we end up with a local (let-bound) function in this particular case (which we didn't get in the fall-through version of `both`). This is then the error you eventually see, a local (let-bound) function is applied, and the netlist generator is going to look for this function in the global namespace, because it operates under the assumption that all local functions have been eliminated.

Does it make sense?

So, having seen this bug I know what to do to fix it: if see any form of error as the subject/scrutinee of a case-expression, I replace that entire case-expression by the error. I'll fix it tomorrow when I'm back in the office. Of course, this doesn't fix the problem that CLaSH cannot handle functions stored inside a register, for that we'll still need a proper defunctionalisation pass.

commit 3a83f2f6 (2016-01-12): caseCon: case of error reduces to error

Fixes #109
closed 2016-01-12 by christiaanb