clash-lang/clash-compiler #111: Exception: CLaSH.Netlist(179): Not in normal form: RHS of case-projection is not a variable

ggreif, 2016-01-12:
Attached [CAM.txt](https://github.com/clash-lang/clash-compiler/files/87788/CAM.txt) exposes another bug (?) in the VHDL generator:
```
*CAM> :vhdl
[1 of 1] Compiling CAM              ( CAM.hs, CAM.o )
Loading dependencies took 0.46461s
Applied 365 transformations
Normalisation took 0.803514s
*** Exception: CLaSH.Netlist(179): Not in normal form: RHS of case-projection is not a variable: case CLaSH.Prelude.BlockRam.blockRam#
     @(CLaSH.Signal.Internal.Clk system 1000)
     @(GHC.TypeLits.^ 2 6)
     @(CAM.Bucket GHC.Types.Bool GHC.Types.Bool)
       64
       CLaSH.Signal.Explicit.systemClock1913188173
       (CLaSH.Sized.Vector.replicate
        @(GHC.TypeLits.^ 2 6)
        @(CLaSH.Sized.Vector.Vec
            2 (GHC.Base.Maybe (GHC.Tuple.(,) GHC.Types.Bool GHC.Types.Bool)))
          (CLaSH.Promoted.Nat.SNat
           @(GHC.TypeLits.^ 2 6)
             64
             (Data.Proxy.Proxy
              @GHC.TypeLits.Nat
              @(GHC.TypeLits.^ 2 6)))
          (CLaSH.Sized.Vector.replicate
           @2
           @(GHC.Base.Maybe (GHC.Tuple.(,) GHC.Types.Bool GHC.Types.Bool))
             (CLaSH.Promoted.Nat.SNat
              @2
                2
                (Data.Proxy.Proxy
                 @GHC.TypeLits.Nat
                 @2))
             (GHC.Base.Nothing
              @(GHC.Tuple.(,) GHC.Types.Bool GHC.Types.Bool))))
       bodyVar_61189
       bodyVar_91186
       eta_i31179
       eta_i41180 of
  CLaSH.Signal.Internal.:-
    (ww1 :: CAM.Bucket GHC.Types.Bool GHC.Types.Bool)
    (ww2 :: CLaSH.Signal.Internal.Signal'
              (CLaSH.Signal.Internal.Clk system 1000)
              (CAM.Bucket GHC.Types.Bool GHC.Types.Bool)) ->
    GHC.Prim.(#,#)
    @(CAM.Bucket GHC.Types.Bool GHC.Types.Bool)
    @(CLaSH.Signal.Internal.Signal'
        (CLaSH.Signal.Internal.Clk system 1000)
        (CAM.Bucket GHC.Types.Bool GHC.Types.Bool))
      ww1
      ww2
```
Changing `readNew` to `id` fixes the generation problem, but of course, the semantics suffer.
B.t.w. this passes the Haskell `test`.

Also I manually defunctionalised the code (relative to issue #109) so that `__VOID__` shouldn't appear in the generated vhdl (using `id` instead of `readNew`). Vivado is happy too.

christiaanb, 2016-01-12:
Given that the constructor of the Signal type is exposed, I'm afraid it's related to the annoying CPR problem described here: https://github.com/clash-lang/clash-compiler/blob/master/clash-ghc/src-ghc/CLaSH/GHC/LoadModules.hs#L288

To fix this, I need to figure out why GHC is doing the worker/wrapper and CPR analysis, so I can fix the specific module in the prelude. Perhaps, CLaSH.Prelude.BlockRam needs to be compiled with -O0.

ggreif, 2016-01-12:
Then this is possibly not a bug at all, as I am compiling with
7.11.20150407, which did not have your CPR suppression patch yet.

ggreif, 2016-01-12:
@christiaanb that said, I am not sure QC works with v8.0, so I am wary to upgrade. Also I have no 7.10.x at work :-(

ggreif, 2016-01-12:
@christiaanb forgot to ask, can you repro it?

christiaanb, 2016-01-12:
My internet at home is broken it seems, and my phone/provider doesn't allow tethering, so I won't be able to test it until I get to the office tomorrow morning.

ggreif, 2016-01-12:
@christiaanb no urgency at all 😀

christiaanb, 2016-01-13:
It works, on my machine (using GHC 7.10.2), the output of the VHDL simulator is:
```

christiaanb, 2016-01-13:
Don't close this bug yet though, we might still need to add a `{-# OPTIONS_GHC -fno-cpr-anal #-}` pragma to the `CLaSH.Prelude.BlockRam` when compiling with a newer version of ghc HEAD.

ggreif, 2016-01-13:
Any ETA for clash in conjunction with GHC v8.0 (or HEAD)?

christiaanb, 2016-01-13:
Sadly not, it depends on when the dependencies of the compiler and the prelude are ready. Especially the `singletons` library and its dependencies: they use TH, and as those of us who run HEAD often, TH breaks almost every _minor_ release of GHC.

christiaanb, 2016-01-13:
I have hacked patches for most dependencies on my own machine to test clash on ghc8, but they are probably out of sync. And, they are hacked, so not in any shape to be merged with the official repo, let alone be put on hackage.

ggreif, 2016-01-15:
@christiaanb Okay I tried something different. Rebuilt `clash-prelude` with `cabal install clash-prelude/ --ghc-option=-fcpr-off` (after thoroughly cleaning it). Did the same to the other packages too. Then generated `:vhdl`. This time I got:
```
*CAM> :vhdl
[1 of 1] Compiling CAM              ( CAM.hs, CAM.o )
Loading dependencies took 0.294112s
Applied 361 transformations
Normalisation took 1.138219s
*** Exception: CLaSH.Netlist(124): CLaSH.Netlist.Util(53): Not in normal from: no Letrec:
 (eta :: CLaSH.Signal.Internal.Signal'
           (CLaSH.Signal.Internal.Clk system 1000)
           (CAM.Out GHC.Types.Bool GHC.Types.Bool)) ->
( a ->
  b ->
  clk ->
  (f :: CLaSH.Signal.Internal.Signal' clk a
        -> CLaSH.Signal.Internal.Signal' clk b) ->
  (x :: CLaSH.Signal.Internal.Signal' clk a) ->
 f x)
@(CAM.Out GHC.Types.Bool GHC.Types.Bool)
@(CAM.Func (CAM.Out GHC.Types.Bool GHC.Types.Bool))
@(CLaSH.Signal.Internal.Clk system 1000)
  ( (ds :: CAM.Out GHC.Types.Bool GHC.Types.Bool) ->
   case ds of
     CAM.None ->
        (ds1 :: CLaSH.Sized.Vector.Vec
                  2
                  (GHC.Base.Maybe (GHC.Tuple.(,) GHC.Types.Bool GHC.Types.Bool))) ->
       GHC.Base.Nothing
       @GHC.Types.Bool
     CAM.FindCol
       (k :: GHC.Types.Bool) ->
        (w2 :: CLaSH.Sized.Vector.Vec
                 (GHC.TypeLits.+ 1 1)
                 (GHC.Base.Maybe (GHC.Tuple.(,) GHC.Types.Bool GHC.Types.Bool))) ->
       CAM.$wa21912645335
       @GHC.Types.Bool
       @GHC.Types.Bool
         GHC.Classes.$fEqBool1912645324
         k
         w2)
  eta
Which has type:
CLaSH.Signal.Internal.Signal'
  (CLaSH.Signal.Internal.Clk system 1000)
  (CAM.Out GHC.Types.Bool GHC.Types.Bool)
-> CLaSH.Signal.Internal.Signal'
     (CLaSH.Signal.Internal.Clk system 1000)
     (CAM.Func (CAM.Out GHC.Types.Bool GHC.Types.Bool))
```

Did I miss something?
Interestingly, replacing `readNew` with `id` still causes this new error, so I definitely made some progress.

christiaanb, 2016-01-15:
I'm observing the same thing. What's weirder is that commenting out the `expectedOutput` function makes the design compile. I don't know yet what's going on here.

christiaanb, 2016-01-15:
OK, I know what's going on here. It's a combination of:
- Me throwing away GHC Core's `Cast` constructs.
- Treating `Signal'` as a "transparent" type, but, as it seems, not in all locations within the compiler
- Slight change in the Core that I'm getting from GHC

So yeah, definitely a bug, slight variations of GHC's Core for the same Haskell descriptions shouldn't influence whether CLaSH can generate VHDL/Verilog or not. I will fix.

christiaanb, 2016-01-15:
As suggested by you already, can you perhaps file this as a separate bug report?

ggreif, 2016-01-15:
Sure.

ggreif, 2016-01-15:
ooops there was still a "don't close yet" comment from you...

christiaanb, 2016-04-18:
Both CAM.hs files compile correctly on GHC8-rc3 (+https://phabricator.haskell.org/D2108)

closed 2016-04-18 by christiaanb