clash-lang/clash-compiler #112: Slow compilation and high memory usage with simple example

adamwalker, 2016-01-15:
Hi,

Running clash --vhdl on the unrolled CRC code below takes 1.5 minutes and uses upwards of 6 gigs of ram. This is unexpected as other examples with similar complexity take a few seconds. 

https://gist.github.com/adamwalker/a0b42571688b05a28058
I tried adding NOINLINE pragmas as was suggested in other bug reports relating to memory usage but this did not help.
Perhaps, as a workaround, there is a more idiomatic way that I can write this to make it faster?

Thanks

christiaanb, 2016-01-15:
As a work around, you can write `generateCRCBit` like so:

```haskell
generateCRCBit :: KnownNat (n+1) => Vec (n + 1) Int -> Bit
generateCRCBit = reduceXor . v2bv . map (theBV !!)
```

There's specialised VHDL/Verilog for the `reduceXor` operator, which makes code-generation faster.

The "problem" in your original design is the `fold` function, which we can compile quickly for vectors that have a length that is a power of 2, but, as is seems, slowly for vector that are not a power of two. The reason for this is that we can generate specialised VHDL/Verilog code for power-of-2 vector, where we cannot do so for other lengths. As a result, we are forced to unroll the definition of fold in-place, which is taking a long time and is apparently consuming a lot of memory as well. 
 
I'll look into it, but am making no promises on a fast fix.

adamwalker, 2016-01-15:
That makes sense. Using foldl1 instead of fold seems to work well too.

christiaanb, 2016-01-15:
`foldl` (and hence `foldl1`) are indeed faster than `fold`, for the same reason that `reduceXor` is faster: we have specialised VHDL/Verilog templates for `foldl`. The "problem" with `foldl` that it creates a circuit that has a long chain of `xor`s of length _n_, where `fold` creates a nice tree of `xor`s of depth _log(n)_. The `reduceXor` function also results in a nice tree of `xor`s.

However, it might be that the VHDL synthesis tool will convert the long chain of `xor`s created by `foldl1` into a nice tree.

adamwalker, 2016-01-15:
Yep, I'm almost certain that it will. Pretty sure I've seen it do that before with some other associative operation.

alex-mckenna, 2022-02-25:
I forgot to comment: I tried this a few days ago and got the following with `clash --vhdl`

```
Clash: Applied 39 transformations
Clash: Normalization took 4.228s
Clash: Netlist generation took 0.022s
Clash: Compiling CRC.topEntity took 4.545s
Clash: Total compilation took 6.861s
```

Which is by no means incredible performance, but I think enough to close this issue.

closed 2022-02-25 by alex-mckenna