clash-lang/clash-compiler #116: Compiler error matching blockram template

adamwalker, 2016-01-18:
This file: https://gist.github.com/adamwalker/01ac3010efb1001b4974

fails to compile with this message:

<no location info>:
    CLaSH Error:
CLaSH.Netlist.BlackBox(84): 
Can't match template for "CLaSH.Prelude.BlockRam.blockRam#" :
"-- blockRam begin\nblockRam_~COMPNAME_~SYM[0] : block\n  signal RAM_~SYM[1]  : ~TYP[2] := ~LIT[2];\n  signal dout_~SYM[2] : ~TYP[6];\n  signal wr_~SYM[3]   : integer range 0 to ~LIT[0] - 1;\n  signal rd_~SYM[4]   : integer range 0 to ~LIT[0] - 1;\nbegin\n  wr_~SYM[3] <= to_integer(~ARG[3])\n  -- pragma translate_off\n                mod ~LIT[0]\n  -- pragma translate_on\n                ;\n\n  rd_~SYM[4] <= to_integer(~ARG[4])\n  -- pragma translate_off\n                mod ~LIT[0]\n  -- pragma translate_on\n                ;\n\n  blockRam_sync : process(~CLK[1])\n  begin\n    if rising_edge(~CLK[1]) then\n      if ~ARG[5] then\n        RAM_~SYM[1](wr_~SYM[3]) <= ~ARG[6];\n      end if;\n      dout_~SYM[2] <= RAM_~SYM[1](rd_~SYM[4]);\n    end if;\n  end process;\n\n  ~RESULT <= dout_~SYM[2];\nend block;\n-- blockRam end"
with context:
Context {bbResult = (Right (Identifier "tmp_32" Nothing,("system",1000)),Unsigned 129), bbInputs = [(Left (Literal (Just (Signed 64,64)) (NumLit 65536)),Signed 64,True),(Right (Identifier "__VOID__" Nothing,("system",1000)),Void,False),(Left (Identifier "repANF_5" Nothing),Vector 65536 (Unsigned 129),False),(Left (Identifier "bodyVar_6" Nothing),Signed 64,False),(Left (Identifier "bodyVar_7" Nothing),Signed 64,False),(Right (Identifier "eta_i1" Nothing,("system",1000)),Bool,False),(Left (Identifier "repANF_8" Nothing),Unsigned 129,False)], bbFunctions = fromList []}
given errors:
[]

christiaanb, 2016-01-18:
As a workaround, instead of using `undefined` in:

```haskell
--Specialise to (Unsigned 64) key, (Unsigned 64) value, (Unsigned 16) address space
topEntity :: Signal Bool -> Signal (Unsigned 16) -> Signal (HashEntry (Unsigned 64) (Unsigned 64)) -> Signal (Maybe (Unsigned 64)) -> (Signal (Maybe (Maybe (Unsigned 64))))
topEntity = top (unpack . crc' . pack) undefined
    where
    crc' :: BitVector 64 -> BitVector 16
    crc' x = slice d15 d0 x
```

use an actual `HashEntry` value like so: 

```haskell
--Specialise to (Unsigned 64) key, (Unsigned 64) value, (Unsigned 16) address space
topEntity :: Signal Bool -> Signal (Unsigned 16) -> Signal (HashEntry (Unsigned 64) (Unsigned 64)) -> Signal (Maybe (Unsigned 64)) -> (Signal (Maybe (Maybe (Unsigned 64))))
topEntity = top (unpack . crc' . pack) (HashEntry False 0 0)
    where
    crc' :: BitVector 64 -> BitVector 16
    crc' x = slice d15 d0 x
```

or any other initial/default value of `HashEntry` that would work for your circuit definition.

As you see, I've fixed the bug in master, so it will be part of the next release, which will probably happen within 2 weeks.

closed 2016-01-18 by christiaanb

adamwalker, 2016-01-18:
Awesome, thanks!

commit 0dd23a8b (2016-01-18): `case undefined of ...` reduces to `undefined`

Fixes #116