clash-lang/clash-compiler #117: No synthesis due to recursive 'go'?

ggreif, 2016-01-22:
'go' is definitely not recursive:
https://raw.githubusercontent.com/ggreif/clash-ground/8b8ba99bfaed927a592f98061e29926b46f17acb/Add.hs
I can strip it down if you wish.

ggreif, 2016-01-22:
It is due to the `error` call. My bad.

closed 2016-01-22 by ggreif

christiaanb, 2016-01-22:
Is unlikely due to `error`, it is much more likely due to `show`. The only strings that the CLaSH compiler can currently handle are string literals. So `"Hello World!"` is OK, but `"Hello " Data.List.++ "World!"` already is _not_.

ggreif, 2016-01-22:
What VHDL would `error` generate then? Undefined values?

christiaanb, 2016-01-22:
Yes, 'X'. I might change this to don't care, '-', in the future.

ggreif, 2016-01-22:
What would it take to transform `Debug.Trace.traceShowId` (and friends) to `id` in the reducer? Can you detect the `unsafePerformIO $ traceIO ...` construct?

christiaanb, 2016-01-22:
Given that `trace` is marked `NOINLINE` according to https://hackage.haskell.org/package/base-4.8.1.0/docs/src/Debug.Trace.html#trace it should be extremely easy. You can try using this black box definition:

`Debug.Trace.json`:
```json
[ { "BlackBox" :
    { "name"      : "Debug.Trace.trace"
    , "type"      : "trace :: String -> a -> a"
    , "templateE" : "~ARG[1]"
    }
  }
]
```

See http://hackage.haskell.org/package/clash-prelude-0.10.5/docs/CLaSH-Tutorial.html#g:14 for more info about adding your own primitives. If the above works, I can add it to the next release of CLaSH.

ggreif, 2016-01-22:
I'll try this later today :-) Thanks!

ggreif, 2016-01-22:
@christiaanb I tried this, did not work:
```
Result size of Tidy Core
  = {terms: 1,400, types: 1,587, coercions: 433}
Loading dependencies took 0.42237s

<no location info>: CLaSH Error:
Debug.Trace
*** Deleting temp files:
```
I used
``` haskell
go a = (trace "hhhh" a, Nothing)
```

ggreif, 2016-01-22:
@christiaanb So omitting `~ARG[0]` from the template will mean that the string argument will be completely ignored?

ggreif, 2016-01-22:
I have to correct myself. This works:
``` haskell
go a = (trace "hhhh" a, Nothing)
```
but this doesn't:
``` haskell
go a = (trace (show a) a, Nothing)
```
So the first argument still gets normalised (I guess it shouldn't).

christiaanb, 2016-01-22:
What's the exact error that you're getting?

christiaanb, 2016-01-22:
And could you give me the complete file again?

ggreif, 2016-01-22:
Added all info to PR #119. Thanks for looking into this!