clash-lang/clash-compiler #120: Better 'Maybe' representation

ggreif, 2016-01-23:
When I have a 3-constructor type T3 then Maybe T3 can be represented as a 4-constructor type. This means 2 bits for the variant logic instead of 3. There are many more examples.

christiaanb, 2016-01-23:
Here's some places to look if you want to make a patch yourself:

* https://github.com/clash-lang/clash-compiler/blob/v0.6.8/clash-lib/src/CLaSH/Netlist/Util.hs#L164

Ericson2314, 2016-01-25:
Note that this isn't *wholly* an improvement because then constructing and eliminating the `Maybe` in `Maybe T3` takes more work.

ggreif, 2016-01-25:
@Ericson2314 yes, it remains to see whether it is a win, as it complicates the encoding/decoding logic and worsens timing. But if there is a case switch on `T3` already then it may come (almost) free.

christiaanb, 2016-01-25:
Due note that while Haskell has nested patterns, Core does not. So while in haskell you might have:

```haskell
case d of
  T1 x -> ...
  T2 y -> ...
  T3 (Just a) -> ..
  T3 Nothing -> ..
```

in Core this will become:

```haskell
case d of
  T1 x -> ...
  T2 y -> ...
  T3 dm -> case dm of
    Just a -> ...
    Nothing -> ...
```

which in turn becomes a 2-multiplexer feeding a 3-multiplexer. Turning this into a 4-multiplexer is more work than just changing the encoding.

martijnbastiaan, 2018-08-08:
Custom bit representations have been added in https://github.com/clash-lang/clash-prelude/pull/168 and https://github.com/clash-lang/clash-compiler/pull/339. As @Ericson2314 mentioned, eliminating `Maybe`s is not an optimization *per se* so it's up to the user to provide encodings. Clash can also automatically derive derivations for simple cases.

closed 2018-08-08 by martijnbastiaan