clash-lang/clash-compiler #123: SaturatingMode::SatWrap can not be synthesized for Fixed

ra1u, 2016-02-06:
for example

     topEntity:: (SFixed 2 6) -> (SFixed 2 6) -> (SFixed 2 6)
     topEntity = satPlus SatWrap 

Works emulated in interpreter, but yields error when synthesised

    <no location info>:
        CLaSH Error:
    CLaSH.Normalize.Transformations(220): Report as bug: caseCon error: case CLaSH.Class.Num.SatWrap of
      CLaSH.Class.Num.SatBound ->
        CLaSH.Sized.Internal.Signed.minBound#
        @8
          8
      CLaSH.Class.Num.SatZero -> lvl5303
      CLaSH.Class.Num.SatSymmetric -> lvl8319

This code get synthesised with other `SaturatingMode`
that is  `SatBound` , `SatZero` and  `SatSymmetric`

Workaround for me was

    plusWraped:: NumSFixedC n1 n2 => SFixed n1 n2 -> SFixed n1 n2 -> SFixed n1 n2
    plusWraped a b = Fixed ((unSF a) + (unSF b))

christiaanb, 2016-02-07:
Glad you found a work-around. I will fix this when I get back in the office on monday.

ra1u, 2016-02-07:
Thank you for your fast and promptly support.

commit 9ceee806 (2016-02-08): `caseCon` should handle non-exhaustive case-expressions

Fixes #123
closed 2016-02-08 by christiaanb