clash-lang/clash-compiler #125: clash doesn't respect -outputdir option

expipiplus1, 2016-02-08:
As far as I can tell the `vhdl` directory is always created in the current working directory. It would be handy to be able to redirect this output from the command line.

christiaanb, 2016-02-08:
I'll see about adding this feature to the next release, which will probably within 1.5-2 weeks.

christiaanb, 2016-02-09:
@expipiplus1 One question though: currently, files end up in `vdhl/<ModuleName>`; with your proposed `-outputdir` option, do you want files to end up in `outputdir/vhdl/<ModuleName>`, or in `outputdir`, or something else?

expipiplus1, 2016-02-09:
I was expecting things to end up in `outputdir/vhdl/<ModuleName>`

christiaanb, 2016-02-10:
So, in the  next version of CLaSH we will have:

  * hdl files can be written to a directory (set by the `-clash-hdldir` flag) other than the current working directory.
    Also respects the `-outputdir` directory, _unless_:
      * `-clash-hdldir` is set to a different directory.
      * `-hidir`, `-stubdir`, and `-dumbdir` are not the same directory as `-odir`

commit 10639325 (2016-02-10): hdl output dir is configurable

Closes #125
closed 2016-02-10 by christiaanb