clash-lang/clash-compiler #126: Irrefutable pattern failed error

ra1u, 2016-02-09:
module MovingAvg where
    import CLaSH.Prelude

    windowN :: (Default a,KnownNat n) => SNat (n+1) -> Signal a -> Vec (n + 1) (Signal a)
    windowN size = window
    movingAvarageNaive size signal =  fold (+) <$> bundle (windowN size signal)

    topEntity :: Signal (Signed 9) -> Signal (Signed 9)
    topEntity = movingAvarageNaive d5

Fails at synthesis with

    [1 of 1] Compiling MovingAvg        ( movingAvarege.hs, movingAvarege.o )
    Loading dependencies took 0.259644s
    clash: panic! (the 'impossible' happened)
      (GHC version 7.10.3 for x86_64-unknown-linux):
        src/CLaSH/Normalize/PrimitiveReductions.hs:250:9-46: Irrefutable pattern failed for pattern [_, consCon]

    Please report this as a GHC bug:  http://www.haskell.org/ghc/reportabug

christiaanb, 2016-02-09:
I think I know what's going on here, I'll fix it when I get back in the office tomorrow morning. Thanks for the bug report.

christiaanb, 2016-02-10:
As a temporary work-around, using either `foldr1` or `fold1` would work.

closed 2016-02-10 by christiaanb

christiaanb, 2016-02-10:
I just released version 0.6.10 of CLaSH which includes this bug fix.

ra1u, 2016-02-10:
Great, I will play around with this when I get home.

commit b9a71c6b (2016-02-10): Primitive reductions should look through `Signal`

Fixes #126