clash-lang/clash-compiler #128: Maintain variable names post-compilation

kybaeus, 2016-02-25:
I would really appreciate a feature to maintain variable naming post-compilation to VHDL or Verilog like MyHDL.

christiaanb, 2016-02-29:
I'm currently working on fixing blockRam inference for Xilinx Vivado. Afterwards, I'll take a look at this.

christiaanb, 2016-03-02:
I'm making progress on this. Some notes:

* Given the following Haskell written by the user:

  ```haskell
  f (x,y) = ...
  ```

  I get the following code from the GHC Haskell compiler:

  ```haskell
  f ds = case ds of (x,y) -> ...
  ```

  which I translate to a component `f` with a single input port called `ds`, and not, as one might expect, a component `f` with two inputs ports `x` and `y`. The problem is, is that I cannot distinguish between `f ds = case ds of (x,y) -> ...` written by the user, or `f ds = case ds of (x,y) -> ...` generated by the GHC Haskell compiler. I don't plan to add any heuristics as to what might be generated or not, so I will always create the single input-port version of the component `f`.

* GHC generates a lot of so-called _worker/wrapper_ function pairs, where you have the user-written function `f`, which becomes the _wrapper_, and the compiler optimized function `f'`, the _worker_. Now, most of the "optimized" functions are actually only running faster when they would be executed on a normal CPU, when translating them to circuits, it makes absolutely no difference. Also, the worker/wrapper transformation does some name-mangling on the argument names, meaning that it is currently sometimes hard to preserve names in the HDL output. However, once GHC8 is released, and I can base CLaSH on top of GHC8, I can finally disable the worker/wrapper transformation, and hence no longer have the annoying name-mangling.

christiaanb, 2016-03-11:
Some of the name-preservation improvements are part of the 0.6.11 release of CLaSH. I'll keep this issue open, as I'm sure we can do better than what's currently implemented.

christiaanb, 2020-01-16:
Clash preserves most names these days; and when it doesn't you simply add a `{-# NOINLINE xyz #-}` pragma

closed 2020-01-16 by christiaanb