clash-lang/clash-compiler #129: Suggestion: make clash-prelude a git submodule

ggreif, 2016-03-04:
I am just letting the idea balloon loose...

`clash-compiler` is depending on `clash-prelude` pretty heavily with respect to semantics, so why not link them in `git` also?

Would you accept a pull request?

christiaanb, 2016-03-04:
I would, but I'm unfamiliar with the semantics of git submodules. So the pull-request should also update the `.travis.yml` and `stack.yaml` accordingly. And preferably you could see to update the [wiki page](https://github.com/clash-lang/clash-compiler/wiki/Getting-started) on how to get started on hacking on CLaSH also then?

ggreif, 2016-03-04:
Might take some rounds, but we'll get there :-)

ggreif, 2016-03-04:
I made a first draft in pull request #130. Let's see whether it tests.

christiaanb, 2016-03-05:
#130 is merged

closed 2016-03-05 by christiaanb

ggreif, 2016-03-05:
Great, I have updated the wiki!