clash-lang/clash-compiler #132: 0.6.11 CLaSH.Netlist(376): Not in normal form: application of a Let/Lam/Case

masterdezign, 2016-03-14:
Hello! The newest CLaSH version introduces a strange error message, which was not the case in the previous versions.

<no location info>:
    CLaSH Error:
CLaSH.Netlist(376): Not in normal form: application of a Let/Lam/Case: (λ(w3 :: CLaSH.Signal.Internal.Signal'
           (CLaSH.Signal.Internal.Clk system 72)
           (GHC.Tuple.(,,)
              (GHC.Tuple.(,)
                 (CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 3 13)
                 (CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 3 13))
              (CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 3 13)
              (CLaSH.Sized.Fixed.Fixed
                 CLaSH.Sized.Internal.Signed.Signed 3 13))) ->
 CLaSH.Prelude.Mealy.$wmealy'2075
   (GHC.Tuple.(,,,,)
    @(GHC.Tuple.(,)
        (CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 3 13)
        (CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 3 13))
    @(GHC.Tuple.(,)
        (CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 3 13)
        (CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 3 13))
    @(GHC.Tuple.(,)
        (CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 3 13)
        (CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 3 13))
    @(GHC.Tuple.(,)
        (CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 3 13)
        (CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 3 13))
    @(GHC.Tuple.(,)
        (CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 3 13)
        (CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 3 13))
      (GHC.Tuple.(,)
       @(CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 3 13)
       @(CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 3 13)
         (CLaSH.Sized.Internal.Signed.fromInteger#
          @16
            16
            0)
         (CLaSH.Sized.Internal.Signed.fromInteger#
          @16
            16
            0))
      (GHC.Tuple.(,)
       @(CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 3 13)
       @(CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 3 13)
         (CLaSH.Sized.Internal.Signed.fromInteger#
          @16
            16
            0)
         (CLaSH.Sized.Internal.Signed.fromInteger#
          @16
            16
            0))
      (GHC.Tuple.(,)
       @(CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 3 13)
       @(CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 3 13)
         (CLaSH.Sized.Internal.Signed.fromInteger#
          @16
            16
            0)
         (CLaSH.Sized.Internal.Signed.fromInteger#
          @16
            16
            0))
      (GHC.Tuple.(,)
       @(CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 3 13)
       @(CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 3 13)
         (CLaSH.Sized.Internal.Signed.fromInteger#
          @16
            16
            0)
         (CLaSH.Sized.Internal.Signed.fromInteger#
          @16
            16
            0))
      (GHC.Tuple.(,)
       @(CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 3 13)
       @(CLaSH.Sized.Fixed.Fixed CLaSH.Sized.Internal.Signed.Signed 3 13)
         (CLaSH.Sized.Internal.Signed.fromInteger#
          @16
            16
            0)
         (CLaSH.Sized.Internal.Signed.fromInteger#
          @16
            16
            0)))
   w3)
  result_010455

christiaanb, 2016-03-14:
Is there any way you could provide the source? It is hard to determine what might be causing this.

christiaanb, 2016-03-14:
I doesnt have to be a reduced test case btw. There is only one thing that I changed in version 0.6.11 that I think could be the cause of this error, but I have no idea why. My guess is, is that this change exposed another error already present for a long time.

masterdezign, 2016-03-14:
Is private mail OK?

christiaanb, 2016-03-14:
Yes, private email is OK, I will even sign an NDA if required.

closed 2016-03-14 by christiaanb

christiaanb, 2016-03-14:
I just released version 0.6.12, which contains the above fix.

commit ce31a8ff (2016-03-14): etaTL: move lambda's through let-bindings

After moving etaTL after application propagation (in order to preserve
more Haskell names), we sometimes ended up with a lamba in the body
of a top-level let-binding, where the argument binder of the
enclosing lambda was not propagated because we moved eta-expansion
after application propagation.

What we do now is ensure that all (created) lamba's in the body
of a let-binder are move upward.

Fixes #132