clash-lang/clash-compiler #134: Complex floats not supported?

ggreif, 2016-03-17:
``` haskell
import Data.Complex
import CLaSH.Prelude

topEntity :: Signal (Complex Float)
topEntity = signal (1 :+ 0)
```
I get:
```
Loading dependencies took 0.189004s
Applied 0 transformations
Normalisation took 0.000317s
*** Exception: CLaSH.Netlist(124): CLaSH.Netlist.Util(53): Not in normal from: no Letrec:
(?a ->
 ?clk ->
 ?(x :: CLaSH.Signal.Internal.Signal' clk a) ->
 x)
@(Data.Complex.Complex GHC.Types.Float)
@(CLaSH.Signal.Internal.Clk system 1000)
  (Data.Complex.:+
   @GHC.Types.Float
     (GHC.Types.F#
        1 % 1)
     (GHC.Types.F#
        0 % 1))
Which has type:
CLaSH.Signal.Internal.Signal'
  (CLaSH.Signal.Internal.Clk system 1000)
  (Data.Complex.Complex GHC.Types.Float)
```

christiaanb, 2016-03-17:
Float and Double are not supported, so neither is Complex Float. See also http://hackage.haskell.org/package/clash-prelude-0.10.6/docs/CLaSH-Tutorial.html#g:19

ggreif, 2016-03-17:
I suspected that this is the case, but the error message made me think that there is "just" an internal hiccup.

Next question, what about fixed-point floats? I keep hearing from colleagues that DSP slices can do this and that VHDL supports it as well. http://www.eda.org/fphdl/fpfaq.html

christiaanb, 2016-03-17:
There are no such things as fixed floats... I think. There is the fixed point type in clash: http://hackage.haskell.org/package/clash-prelude-0.10.6/docs/CLaSH-Sized-Fixed.html for if you need fractional numbers. I prefer not to use the VHDL fixed/float package, because that is VHDL-2008, and also there is no equivalent in (System)Verilog.

christiaanb, 2016-03-17:
Also, and I only know this because my friend builds floating point adders, the operations in the VHDL float package don't fully adhere to the IEEE 754 standard.

ggreif, 2016-03-17:
Thanks, I think I am fully happy with `SFixed int frac` for now.

ggreif, 2016-03-17:
Heh, of course I meant *fixed-point reals* above :-)

mgajda, 2016-05-21:
One could think into making a floating point core library, just like one in Oberon system:
http://www.xilinx.com/support/documentation/xcell_articles/91-5-oberon-system-implemented-on-a-low-cost-fpga-board.pdf
The problem here is that multiplier uses variable number of clock cycles in order to maintain highest possible frequency.

martijnbastiaan, 2019-02-22:
This question has been answered, so I'm closing this report. Thanks for the question :).

We should develop some infrastructure to cope with components using variable number of clock cycles, like @mgajda notes, but that's for another issue/enhancement report.

closed 2019-02-22 by martijnbastiaan