clash-lang/clash-compiler #135: No testbench generated

ggreif, 2016-03-18:
With a somewhat older `clash`, namely
```
$ ~/.cabal/bin/clash --version
CAES Language for Synchronous Hardware, version 0.6.8 (using clash-lib, version: 0.6.8)
```
I observed that a `topEntity` with a non-function signature
``` haskell
topEntity :: Signal (SFixed 2 10)
```
will not get a testbench generated even if I have
``` haskell
expectedOutput :: Signal (SFixed 2 10) -> Signal Bool
expectedOutput = outputVerifier $(v ...)
```
available. I tricked it into submission by giving `topEntity` a dummy input signal, but ignoring it:
``` haskell
topEntity :: Signal Bool -> Signal (SFixed 2 10)
topEntity _ = ... as before ...
```
For
```
CAES Language for Synchronous Hardware, version 0.6.15 (using clash-lib, version
: 0.6.13)
```
this trick does not work any more. What can I do?

ggreif, 2016-03-18:
I ended up always inputting a `False` signal and using `mux`. But this is not very satisfactory :-(

christiaanb, 2016-03-21:
Test benches will now also be generated for circuits that have no inputs. I'll release a new version of clash containing this feature today.

commit 7ec8fe4c (2016-03-21): Generate test bench for circuits with no inputs

Fixes #135
closed 2016-03-21 by christiaanb