clash-lang/clash-compiler #136: ghdl fails on some tests

CodiePP, 2016-03-25:
when running the testsuite, ghdl complains about a file not found. 
I compiled the newest ghdl (from GitHub), version 5.2.1

How can I trace this test case? 
run ghdl manually to find the problem?

Here the message from the testsuite:
>    VHDL & Verilog
      Calculator
        CLaSH(VHDL):      OK (4.13s)
          Applied 203 transformations
          Applied 58 transformations
          Applied 99 transformations
        GHDL (import):    OK
        GHDL (make):      OK (0.10s)
        GHDL (sim):       FAIL
          Program /usr/local/bin/ghdl failed with code 1
           Stderr was: 
          /usr/local/bin/ghdl: cannot find entity or configuration calculator_testbench
>          
>           Stdout was:

christiaanb, 2016-03-25:
When you're in the `clash-compiler` directory do:

- `stack exec clash -- --vhdl -iexamples examples/Calculator/hs`
- `cd vhdl/Calculator`
- `mkdir work`
- `ghdl -i --workdir=work *.vhdl`
- `ghdl -m --workdir=work calculator_testbench`
- `ghdl -r calculator_testbench`

What supprises me, is that `GHDL (make)` succeeds in your code, so it must've produced something....

CodiePP, 2016-03-26:
Here the complete messages of a run of 

> stack exec -- clash-testsuite

[output_2016-03-26.txt](https://github.com/clash-lang/clash-compiler/files/190421/output_2016-03-26.txt)

christiaanb, 2016-03-27:
ok, most of those errors just seem to be bugs in GHDL. I will test all the failing test-cases in modelsim just to be sure. Also GHDL 0.31 handles the current testsuite just fine, it seems there are some major regressions on GHDL 0.34dev.

christiaanb, 2016-03-29:
OK, I tested everything in modelsim, and it all works. My conclusion is that GHDL 0.34dev is seriously broken. I suggest to try the officially released version, v0.33: https://github.com/tgingold/ghdl/releases/tag/v0.33

leonschoorl, 2019-01-04:
(because I could find a suitable binary for my OS)
I just tested with my own build from ghdl's current master:
```
$ ghdl --version
GHDL 0.36-dev (20181129-140-g591f3ffe) [Dunoon edition]
 Compiled with GNAT Version: 7.3.0
 llvm code generator
```

And all VHDL tests pass

martijnbastiaan, 2019-01-07:
To install GHDL this way, execute:

```bash
#!/bin/bash
set -e

VERSION=20181129
PREFIX=/usr/local

sudo apt install llvm clang gnat zlib1g-dev build-essentials

wget "https://github.com/ghdl/ghdl/archive/$VERSION.tar.gz"
tar xzf $VERSION.tar.gz
cd ghdl-$VERSION
mkdir build
cd build
../configure --with-llvm-config --prefix=$PREFIX
make
sudo make install
```

christiaanb, 2019-01-07:
Guess it's solved

closed 2019-01-07 by christiaanb