clash-lang/clash-compiler #137: missing GHC/TypeLits/Normalise after stack update nightly

flip111, 2016-03-25:
I was trying stack again after it was already installed a while ago using stack. Now when running the exec command i get this error

```
» stack exec --resolver=nightly -- clash --interactive
Run from outside a project, using implicit global project config
Using resolver: nightly specified on command line
Selected resolver: nightly-2016-03-25
Downloaded nightly-2016-03-25 build plan.    
Caching build plan
Fetched package index.                                                                                    
Populated index cache.    
CLaSHi, version 0.6.8 (using clash-lib, version 0.6.8):
http://www.clash-lang.org/  :? for help
<command line>: Could not find module ‘GHC.TypeLits.Normalise’
Use -v to see a list of the files searched for.

» stack exec --resolver=nightly -- clash --interactive -v
Run from outside a project, using implicit global project config
Using resolver: nightly specified on command line
Selected resolver: nightly-2016-03-25
CLaSHi, version 0.6.8 (using clash-lib, version 0.6.8):
http://www.clash-lang.org/  :? for help
Glasgow Haskell Compiler, Version 7.10.3, stage 2 booted by GHC version 7.10.2
Using binary package database: /home/flip111/.stack/programs/x86_64-linux/ghc-7.10.3/lib/ghc-7.10.3/package.conf.d/package.cache
Using binary package database: /home/flip111/.stack/snapshots/x86_64-linux/nightly-2016-03-25/7.10.3/pkgdb/package.cache
Using binary package database: /home/flip111/.stack/global-project/.stack-work/install/x86_64-linux/nightly-2016-03-25/7.10.3/pkgdb/package.cache
wired-in package ghc-prim mapped to ghc-prim-0.4.0.0-6cdc86811872333585fa98756aa7c51e
wired-in package integer-gmp mapped to integer-gmp-1.0.0.0-3c8c40657a9870f5c33be17496806d8d
wired-in package base mapped to base-4.8.2.0-0d6d1084fbc041e1cded9228e80e264d
wired-in package rts mapped to builtin_rts
wired-in package template-haskell mapped to template-haskell-2.10.0.0-3c4cb52230f347282af9b2817f013181
wired-in package ghc mapped to ghc-7.10.3-be5219aceb0976c913043e27d1693295
wired-in package dph-seq not found.
wired-in package dph-par not found.
Hsc static flags: -fcpr-off
Loading package ghc-prim-0.4.0.0 ... linking ... done.
*** gcc:
/usr/bin/gcc -fno-stack-protector -DTABLES_NEXT_TO_CODE -B/home/flip111/.stack/programs/x86_64-linux/ghc-7.10.3/lib/ghc-7.10.3/integ_2aU3IZNMF9a7mQ0OzsZ0dS --print-file-name libgmp.so
Loading package integer-gmp-1.0.0.0 ... linking ... done.
Loading package base-4.8.2.0 ... linking ... done.
wired-in package ghc-prim mapped to ghc-prim-0.4.0.0-6cdc86811872333585fa98756aa7c51e
wired-in package integer-gmp mapped to integer-gmp-1.0.0.0-3c8c40657a9870f5c33be17496806d8d
wired-in package base mapped to base-4.8.2.0-0d6d1084fbc041e1cded9228e80e264d
wired-in package rts mapped to builtin_rts
wired-in package template-haskell mapped to template-haskell-2.10.0.0-3c4cb52230f347282af9b2817f013181
wired-in package ghc mapped to ghc-7.10.3-be5219aceb0976c913043e27d1693295
wired-in package dph-seq not found.
wired-in package dph-par not found.
*** Deleting temp files:
Deleting: 
*** Deleting temp dirs:
Deleting: 
<command line>: Could not find module ‘GHC.TypeLits.Normalise’
Locations searched:
  GHC/TypeLits/Normalise.hs
  GHC/TypeLits/Normalise.lhs
  GHC/TypeLits/Normalise.hsig
  GHC/TypeLits/Normalise.lhsig
```

christiaanb, 2016-03-25:
try `stack install --resolver=nightly ghc-typelits-natnormalise`

flip111, 2016-03-25:
thanks for the help

by the way, this was also required
```
stack install --resolver=nightly ghc-typelits-extra
stack install --resolver=nightly clash-prelude
```

closed 2016-03-25 by flip111