clash-lang/clash-compiler #143: missing upper bounds

hvr, 2016-05-19:
it seems like `clash-compiler` is missing upper bounds which results in install-plans bitrotting over time as can be seen at e.g.
http://matrix.hackage.haskell.org/package/clash-verilog#GHC-7.10/clash-verilog-0.6.5

please add upper bounds uploaded to Hackage as this causes causes unecessary work for Hackage Trustees

ggreif, 2016-05-19:
@hvr just running `cabal gen-bounds` and pasting the results should do it? (http://softwaresimply.blogspot.fr/2015/08/cabal-gen-bounds-easy-generation-of.html)

mgajda, 2016-05-21:
With regards to avoiding bitrot - putting CLaSH into Stackage would help as well.

hvr, 2016-05-21:
@ggreif that would be a good start, I'm sure @mightybyte will be interested to know how it works out :-)

@mgajda afaik, placing a package in Stackage doesn't avoid the bitrot I'm referring to (i.e. compile errors too lax version bounds). Stackage helps more with helping make sure there's a large set of recent package versions co-buildable with each other. In fact, Stackage fights the opposite issue, namely too tight version bounds (resulting in planning failures rather than compile errors).

christiaanb, 2016-06-09:
All the latest released `clash-X` packages now have upper bounds on all of their dependencies.

closed 2016-06-09 by christiaanb