clash-lang/clash-compiler #147: CλaSH FPGA Starter

wrjianbo, 2016-05-26:
I read this tutorial about FPGA http://christiaanb.github.io/posts/clash-fpga-starter/
I tried the code in my laptop. But when I use "clash --vhdl Blinker.hs" command to generate vhdl files, I find something wrong. 
This is what CλaSH should generate:
[image]
However this is what I actually got:
[image]
My system is OS X EI Capitan 10.11.5

christiaanb, 2016-05-26:
What version of CLaSH are you using? (you can check using `clash --version`) The bug you reported should be fixed as of version 0.6.16. If you have an older version you can update to the latest version (0.6.17 at the time of this writing) by:

```
cabal update
cabal install clash-ghc-0.6.17
```

christiaanb, 2016-05-31:
@wrjianbo Are you still experiencing the same problems using clash 0.6.17? or are you having problems installing clash 0.6.17?

wrjianbo, 2016-06-09:
I've updated my clash version and now it works fine, thanks a lot

christiaanb, 2016-06-09:
Thanks for confirming that everything works.

closed 2016-06-09 by christiaanb