clash-lang/clash-compiler #148: Preserve unused top level inputs

expipiplus1, 2016-05-26:
Currently if a top level input is unused (or optimized away) in a function it is not present in the generated VHDL.

Any consumer of this VHDL has to deal with these inputs changing as the logic underneath them changes. With the Xilinx ISE compiler missing ports can lead to some pretty cryptic error messages (type errors?!).

It would be handy if the 'signature' of the top level entity could remain the same, irrespective of what's going on underneath. Of course a warning on unused inputs would go very nicely with this.

christiaanb, 2016-05-26:
Could you provide some example code? It might be that GHC is optimizing it away...

expipiplus1, 2016-05-26:
This was happening on normal inputs with a fairly large testcase on a coworker's computer, I'll see if we can cut it down tomorrow.

An example I can produce now is /similar/; it's the inconsistent exposure of clock ports. In this example no port named `system1000` is created. Though that not being exposed is a little more understandable.

```haskell
{-# ANN foo
  (defTop { t_name = "foo"
          , t_inputs = ["i"]
          , t_outputs = ["o"]
          }
  )#-}
foo :: Signal Bool -> Signal Bool
foo = fmap id
```

expipiplus1, 2016-05-27:
This issue is actually very difficult to reproduce, it /seems/ to be nondeterministic.

This still happens after a 'make clean', ruling out our build system as the culprit.

We can reproduce it by:

- Compile things successfully
- Touch the file with the top level function
- Compile unsuccessfully

or

- Compile things unsuccessfully
- Touch the file with the top level function
- Compile successfully

expipiplus1, 2016-05-27:
It seems to happen about 1% of the time, but once it's happened once it's likely you'll get a string of these failures.

This is what we think is happening, obviously the human brain is good at finding patterns in noise though...

christiaanb, 2016-05-27:
When this happens, do you know if the number of VHDL files changes also? And do you have things like topEntity0.vhdl, topEntity1.vhdl, topEntity2.vhdl?

If so, then I think this is the somewhat strange behaviour of GHCs worker/wrapper transformation in the presence of previously compiled function/modules that I have also sometimes observed. Though I never noticed any inputs disappearing; only more topEntityX.vhdl files appearing.

The next major version (0.7/1.0) of CLaSH, based on GHC8, has the worker/wrapper disabled. Hopefully that will also fix this weird disappearing unused input port situation.

christiaanb, 2016-05-27:
@expipiplus1 Just to be clear, what you are experiencing is given the following Haskell:

```
topEntity :: Int -> Int -> Int
topEntity x y = x
```

You sometimes end up with the following VHDL entity:

```vhdl
entity topEntity is
  port(x : in signed(63 downto 0);
       y : in signed(63 downto 0);
       res : out signed(63 downto 0));
end;
```

and other times:

```vhdl
entity topEntity is
  port(x : in signed(63 downto 0);
       res : out signed(63 downto 0));
end;
```

right?

expipiplus1, 2016-05-27:
Yes, that's exactly what's happening.

Next time it happens I'll make a count of the number of vhdl files.

expipiplus1, 2016-12-04:
This hasn't happened for a while.

closed 2016-12-04 by expipiplus1